SpikeInterface/spikeinterface #2724: Unit locations discrepancy between SI and PHY

taningh86, 2024-04-16:
Hi,
I am seeing some units in Phy that have locations in the shank not where SI is showing when I run `sw.plot_unit_locations(sa,width_cm =2, backend ="ipywidgets")`. I am curious if this is a known problem or something rare and in cases when this happens what is the approach as far as which one to trust?
I am happy to share pictures if needed.

Thanks!

zm711, 2024-04-16:
Pictures would be nice (between Phy and SI) for me. I would like to see what is actually happening :)

taningh86, 2024-04-16:
[image]
[image]
Here you go. examples of one of the many I found.

alejoe91, 2024-04-16:
@taningh86 phy IDs are not SI IDs! The reason is that phy needs integers. You should have a si_unit_id column in the phy cluster view ;)

zm711, 2024-04-16:
This has come up multiple times. Let's not close this until we have a document PR linked where we emphasize that they are not the same :)

taningh86, 2024-04-16:
> @taningh86 phy IDs are not SI IDs! The reason is that phy needs integers. You should have a si_unit_id column in the phy cluster view ;)

Hi @alejoe91 I don't see one. I have attached a pic of it here. This is the only window I see in Phy for clusterview. Is this you are talking about?
[image]

zm711, 2024-04-16:
Do you have `.tsv` file called like `cluster_info.tsv`? If you look in there there should be a call. Maybe it is not displaying?

taningh86, 2024-04-16:
You mean in the Phy folder or the kilosort output folder? I don't see a cluster_info.tsv file in the Phy folder, neither in the sorter_output folder.
[image]

taningh86, 2024-04-16:
But it should be there in the sorter output folder since we need it to assign spiketrain and other info to the right clusters, isn't? Is cluster_group.tsv the same?

zm711, 2024-04-16:
You are just doing this from kilosort natively. Ie you never ran `export_to_phy`?

taningh86, 2024-04-16:
Ah, that may be the issue! yes, I am doing it directly from kilosort. My bad! Let me try that.

zm711, 2024-04-16:
It should still work though. How are you reading the kilosort units back into spikeinterface? `se.read_kilosort` or `se.read_phy`?

taningh86, 2024-04-16:
I was running everything in spikeinterface but recently started using Phy. I also tried the manual curation in spikeinterface. So, I don't think I am reading it back. To read sorted data (which was sorted in docker) I use 'ss.read_sorter_folder(base_folder/"results_KS25").

taningh86, 2024-04-16:
It seems to work after exporting it to phy. Unit locations seems to match. However, correlograms look different. I can't think of a reason why they would change since they are just exported from SI and it should look the same, isn't?

taningh86, 2024-04-16:
Also, the KSLabel column that says 'good' or 'mua' is not there anymore. Is there a way we can still have it?

taningh86, 2024-04-16:
But, still curious why the unit locations or ID's should change, whether I export data from spikeinterface to phy or just run it directly on the kilosort output folder as @zm711 also said.

zm711, 2024-04-16:
Maybe @alejoe91 has a more elegant solution, but you could just copy the info from the `KSlabel.tsv` into a column of the `cluster_info.tsv` and it should display. It's been a while since I've played with exporting to phy... (even just copying the file into the folder should work, but then it might not update if you're mixing and matching curation). Since the KSlabel is just a kilosort thing, it won't be automatically generated by `export_to_phy` since people using other sorters might not have that info :)

Correlograms could look slightly different because the correlograms aren't exported. Phy calculates its own correlograms inside the gui itself. So the way it displays that info would likely be slightly different. So you would have to check how phy calculates to compare to how SI calculates to see what the differences are.

As far as unit locations/IDs I think I would need to think more about it. Would you be able to post your analysis script here so we can see if any operations may have been performed that would change numbering?

taningh86, 2024-04-16:
Sure @zm711 Here is my code. Its pretty similar to the code in the tutorial (oct 2022) until the sorting stage. Then postprocessing codes are mostly from the documentation and slightly modified for my use.

``import spikeinterface as si
import spikeinterface.extractors as se 
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt
import numpy as np
from pathlib import Path
print(si.__version__)
import warnings
warnings.simplefilter("ignore")

%matplotlib widget

import os
%pwd
os.chdir("G:/Cat_GT_Out/catgt_NPX2_2_28_24_2Rght_LHA_RSP_EXP_g0")
%pwd
base_folder = "G:/Cat_GT_Out/catgt_NPX2_2_28_24_2Rght_LHA_RSP_EXP_g0"
recording_spikeglx = se.read_spikeglx(folder_path = base_folder, stream_id = 'imec0.ap', all_annotations = False)
recording_spikeglx
recording_spikeglx.annotate(is_filtered=False)

channel_ids = recording_spikeglx.get_channel_ids()
fs = recording_spikeglx.get_sampling_frequency()
num_chan = recording_spikeglx.get_num_channels()
num_segments = recording_spikeglx.get_num_segments()

recording_to_process = recording_spikeglx
recording_cmr = spre.common_reference(recording_to_process, reference='global', operator='median')
w = sw.plot_timeseries({"raw": recording_to_process, "common": recording_cmr},
                        clim=(-50, 50), time_range=[100, 110], order_channel_by_depth=True,
                        backend="ipywidgets")    #only includes CMR and NO bandpass_filter. So its slightly modified from the code in tutorial

fs = recording_cmr.get_sampling_frequency()
recording_sub = recording_cmr.frame_slice(start_frame=0*fs, end_frame=1821*fs)
recording_sub

job_kwargs = dict(n_jobs=10, chunk_duration="1s", progress_bar=True) #this one I need to understand
base_folder = Path(base_folder)

recording_saved = si.load_extractor(base_folder / "preprocessed")

recording_saved = si.load_extractor(base_folder / "preprocessed")
recording_loaded = si.load_extractor(base_folder / "preprocessed")
w = sw.plot_timeseries({"preprocessed": recording_sub, "saved": recording_saved, "loaded": recording_loaded},
                        clim=(-50, 50), mode="line",
                        backend="ipywidgets")

sorter_params = job_kwargs
sorting_KS25 = ss.run_sorter('kilosort2_5', recording_saved, 
                            output_folder=base_folder / 'results_KS25',
                            verbose=True, docker_image=True, 
                            **sorter_params)

sorting_KS25 = ss.read_sorter_folder(base_folder/"results_KS25")
sorting_KS25
recording_saved = si.load_extractor(base_folder / "preprocessed")
sorting = sorting_KS25
print(sorting)

sorting_analyzer = si.create_sorting_analyzer(
    sorting=sorting, 
    recording=recording_saved, 
    format='binary_folder', 
    folder= 'sorting_analyzer'
)

sa = sorting_analyzer
job_kwargs = dict(n_jobs=8, chunk_duration="1s", progress_bar=True)
sorting_analyzer.compute(
    "random_spikes",
    method="uniform",
    max_spikes_per_unit=500,
)
sa.compute("waveforms", ms_before=1.0, ms_after=2.0, return_scaled=True, **job_kwargs)
sa.compute("templates", operators=["average", "median", "std"])
sa.compute('spike_amplitudes', **job_kwargs)
sa.compute('unit_locations', method="monopolar_triangulation")
sa.compute('spike_locations', method="center_of_mass", **job_kwargs)
sa.compute('correlograms')
sa.compute('template_similarity')
print(sorting_analyzer)``

After this its mostly widgets, template and QM modules.

taningh86, 2024-04-18:
Hi @zm711 Have you ha a chance to look at the code above? I am curious why would SI unit locations be any different than Phy since it works on the kilosort data. I am still finding lot of units with different site locations between the two.

zm711, 2024-04-18:
Well the sites could be slightly different! I want to emphasize that first. You get to choose your `unit_location` functionality (you chose to do `monopolar_triangulation`. so this could shift you slightly. I'm not sure though. let me tag @alejoe91 back into this and see if he has any ideas.

taningh86, 2024-04-18:
Ok. I agree that sites could be slightly different along the same vicinity with the computation I chose. However, I am seeing locations change drastically, about 4000 micron along the shank between SI and Phy. I am happy to share more pictures for comparisons of multiple sites I found that jumped location quite far along the shank.

alejoe91, 2024-04-18:
Did you apply any postprocessing to the la output, e.g. removing redundant units? I agree that this is weird. Can you check that the waveforms/templates look similare between phy and si?

taningh86, 2024-04-18:
Yeah, waveforms looks similar between the units and as far as I know I have not done any removal of redundant units. I am away from the analysis computer currently but will get back later today when i am at it.
P.S I code I shared above is what I did with this dataset. I have done removal of redundant units before with other datasets but have done those always after sorting (mostly to check).

taningh86, 2024-04-19:
Hi @alejoe91 Sorry i could not get access to the analysis computer later half of the day yesterday. I ran Phy again on another dataset and made sure no removal of redundant units were run. I am still getting wide variation in spike loction between Phy and spikeinterface. You guys have never seen this? I need to make sure I am not doing anything wrong. So let me know what y need from my end- more codes or even my dataset.
Also, my usual protocol is to run raw data through CAT-GT and then use the CatGT output in spikeinterface. Then I only run the CMR preprocessing (even though CatGT runs a similar mean based approach) on that data and no bandpass filter since catGT already does that. I believe any of these preprocessing steps that I run in spikeinterface should not change the locations of the unit because they are done all before kilosorting. Let me know whatever you need from my end to diagnose the issue.

zm711, 2024-04-19:
I guess we haven't really confirmed the phy units vs the si units. could you do `se.read_phy` to make your sorting object and then do the sorting_analyzer reading phy and see what that looks like?

So

```python
sorting = se.read_phy(r'path/to/phy/folder')
recording = se.load_extractor(r'path/to/rec')
analyzer= si.create_sorting_analyzer(sorting, recording, xx)
analyzer.compute(xx)
```

taningh86, 2024-04-22:
Hi @zm711 I did what you suggested. below is the code:
``sorting = se.read_phy(r'G:\Cat_GT_Out\catgt_NPX2_2_28_24_2Rght_LHA_RSP_EXP_g0\phy_KS25')
recording_saved = si.load_extractor(base_folder / "preprocessed")
analyzer= si.create_sorting_analyzer(
    sorting=sorting, 
    recording=recording_saved,
    format='binary_folder', 
    folder= 'sorting_analyzer_2'
)
analyzer.compute("waveforms", ms_before=1.0, ms_after=2.0, return_scaled=True, **job_kwargs)
analyzer.compute("templates", operators=["average", "median", "std"])
analyzer.compute('spike_amplitudes', **job_kwargs)
analyzer.compute('unit_locations', method="monopolar_triangulation")
analyzer.compute('spike_locations', method="center_of_mass", **job_kwargs)
analyzer.compute('correlograms')
analyzer.compute('template_similarity')
print(analyzer)``

Unfortunately, the unit location mismatch still exists. Even after creating a sorting object from the Phy folder I was expecting the unit locations in spikeinterface to match exactly with unit locations in Phy. Not sure where the error is originating from - Phy or Spikeinterface. But, whatever the source this is a serious problem in my opinion considering how different conclusions can be made in any given dataset and what the user wants to get out of the data. What do you guys think? Should I raise this issue with Phy also?

taningh86, 2024-04-22:
Is it because of the additional 'monopoar tiangulation' method being used in spikeinterface to compute unit locations? Can spikeinterface just copy what Kilosort found? My impression was that's what spikeinterface was doing.

zm711, 2024-04-23:
I think it would be worth trying `center_of_mass` (which I believe is what phy is approximating). 

As far as Phy that is in maintenance mode. I try to hop over there sometimes to help people with install, but the developer isn't responding to issues any more. So with phy what you see is what you get. Probably better just to troubleshoot here.

taningh86, 2024-04-23:
Ok. I will try center_of_mass and update if the mismatch goes away. Should be able to let you know by afternoon.

taningh86, 2024-04-24:
finally got around it. I ran 'center_of_mass' for spike and unit locations and its still showing the same locations in the probe in spikeinterface as it was with 'monopolar_triangulation'. So spikeinterface holds the same site with both methods and its different from Phy.
The thing that is worrying for me is the degree of deviation between the two. Locations are switched more than 4000 microns along the shank axis but not a few microns. And, almost all discrepancy of unit locations between Phy and SI is in this scale. This seems like a huge deviation and a big issue for me. it would be great if the team can check if they are observing this in their dataset as well. If this is found consistently then its a problem. I am not sure why SI cannot just use the spike and unit locations what kilosort found but has to run its own computation on top of it. I would think that Phy is just copying Kilosort. In case if its not doing that then why run through all that computations in kilosort. Please let me know what you all think.
You guys certainly know about this more than me but I would think my logic here makes sense and we should not be seeing this difference between the wtwo.

zm711, 2024-04-24:
Couple points:

1) SI is not meant to be a Kilosort-Phy pipeline, but rather an agnostic pipeline. Kilosort-Phy were written in the same lab and so were developed together initially. So writing files like the amplitudes was because KS and Phy were able to communicate what type of files they wanted. Because SI works for all sorters (or at least tries to. they are added as they come), it needs to be able to calculate this kind of info. In addition SI gives the user the ability to decide between things like COM vs monopolar triangulation which have time vs accuracy trade-offs. This allows people to have completely customized pipelines. Thus rather than just read the KS output amplitudes etc it calculates its own which is what it does for all sorting information

2) Could you try to take the sorting analyzer and do `export_to_phy` into a separate folder (I don't want you to overwrite your current phy). This will write all the Phy and then we can see if Phy receives the info the SI is seeing if it also has the units in those locations.

taningh86, 2024-04-24:
> Couple points:
>
> 1. SI is not meant to be a Kilosort-Phy pipeline, but rather an agnostic pipeline. Kilosort-Phy were written in the same lab and so were developed together initially. So writing [...]

Ok. I get the advantage of the agnostic pipeline and it makes sense. But what I am slightly confused about is that spikeinterface is doing over what kilosort or other sorters found but not on raw-data itself. correct? if so, then its still dependent on whatever sorter was used. 
I will export the export to phy and will update you. I did it in the past and the unit locations are preserved from Si to Phy. Will update you.

taningh86, 2024-04-24:
> Hi, The 4000 microns diff between phy and spikeinterface must be due to the fact the the channel location are not the same I guess if you run kilosort directly outside from spikeinterface.

@samuelgarcia Are you saying channel location changes depending on if kilosort is ran on its own or wthin spikeinterface? Not clear why that should be and if there's any rationale for that. My understanding is that channel map is fixed by the probe itself and it should not change. Hope I am understanding you right.

zm711, 2024-04-24:
> Ok. I get the advantage of the agnostic pipeline and it makes sense. But what I am slightly confused about is that spikeinterface is doing over what kilosort or other sorters found but not on raw-data itself. correct? if so, then its still dependent on whatever sorter was used.

Basically sorters return a series of spike times and spike identities. Some sorters (like KS) return extra stuff. So SI in this case is going over the raw data with the spike times and identities and calculated a bunch of values you'd be interested in (amplitudes, locations etc). This has the benefit that you can preprocess data (like apply filters etc) on the raw data. Kilosort actually returns amplitudes that aren't real in the sense that they aren't voltages whereas si gives you the option to return real voltages. So there are benefits to this style. Am I answering this question?

>Are you saying channel location changes depending on if kilosort is ran on its own or wthin spikeinterface?

The key here is that if you run kilosort on your own you are making your own channel map. So it is possible to be wrong. The goal of probeinterface is that the channel map can be made automatically for you to reduce the chance of human error + it has the benefit of being open-source so if some spots an error they can fix it for everyone. So if you run kilosort on its own you make your own map whereas if you run kilosort through spikeinterface you trust the spikeinterface map. These could be different although they shouldn't be.

alejoe91, 2024-04-24:
I think that you ran Kilosort via SpikeInterface though, right?

I second Zach's comment: we'd like to be as sorter-agnostic as possible! Every step in spikeinterface is very well documented and reproducible, so we can tell you immediately how the locations/depth are computed. For specific sorters, this information might not be as straight. Honestly, il not sure exactly how KS computes depth. You should check the source code.

For how many units you see such a large difference?

taningh86, 2024-04-24:
So spikeinterface is not using the cluster information sorters come up with after feature extraction and running PCA? Not sure about other sorters but KS outputs cluster information along with channel positions and channel map. But I do understand you about what you meant by agnostic approach more clearly now. I see that KS don't really puts out spike or unit locations besides channel positions and map. Looks like those are two different things. And, may be spikeinterface has a slightly different map then what kilosort has by default and thats why we are seeing this mismatch. However, I am not sure why channel locations/map are different depending on what sorter we choose or manually done. Looks like they are not hard set as I initially thought. I need to understand this better. Could you share some materials that you think has a good in-depth write up on this. NPX manual is not that informative I think.

taningh86, 2024-04-24:
> I think that you ran Kilosort via SpikeInterface though, right?
>
> I second Zach's comment: we'd like to be as sorter-agnostic as possible! Every step in spikeinterface is very well documented and [...]

I see this anywhere fom 10-15% of the units. My main concern is how far apart they are between the two. The mismatch has always been in units which shows in a location in SI but in Phy it shows up 4000 micron above or vice-versa. I could understandably tolerate slight mismatch as long the units shows up in a close region but thats not the case.

zm711, 2024-04-24:
> However, I am not sure why channel locations/map are different depending on what sorter we choose or manually done. 

The probe is a fixed thing. But the headstage and thus the device indices can be different. So contact number 1 could actually be data row 4 in the real data. So when you make your channel map in Kilosort you are setting the channel locations as they relate to the raw data matrix. This is really easy to mess up. So probeinterface takes care of all of that.

```
rec = recording.set_probe(x)
```

Does all that mapping for you under the hood. So that is abstracting away. If you are running kilosort in Spikeinterface then there shouldn't be a discrepancy because spikeinterface is generating the map for you.

> So spikeinterface is not using the cluster information sorters come

What cluster info? KS creates pc features and template features, but spikeinterface calculates that itself, so we don't need to read those. Mountainsort5 for example just returns spike times and unit ids because they know that spikeinterface will handle all other feature generation. So other cluster information is very sorter specific. 

> Could you share some materials that you think has a good in-depth write up on this. NPX manual is not that informative I think.

Have you looked at the [probeinterface docs](https://probeinterface.readthedocs.io/en/main/index.html). That goes into depth what a channel map really means. So I would actually start there to try to understand the difference between contact ids (the numbers on your probe) and device indices (the numbers on your computer).

> I see this anywhere fom 10-15% of the units. 

That's super high. We definitely want to get to the bottom of what is going! I guess I never answered your question, but I have never seen this before. But I also don't use neuropixels. Maybe there is something with how the neuropixel mapping is being read in.... Since I don't use NP I think @alejoe91 would be the better one to think about that portion.

taningh86, 2024-04-24:
Thanks for the explanation! I am getting a better understanding of these behind the scene computations and why. 
``` What cluster info? KS creates pc features and template features, but spikeinterface calculates that itself, so we don't need to read those. Mountainsort5 for example just returns spike times and unit ids because they know that spikeinterface will handle all other feature generation. So other cluster information is very sorter specific.```
By Cluster info I meant `spike_clusters.npy` file that kilosort creates. Doesnt that have the information on what unit is assigned to each cluster? For example, when I am trying to extract spiketrain info from the `spiketimes.npy` file I extract spiketimes of a specific unit using the info in spike_clusters.npy.
And, to answer both @alejoe91 and your question, I am running kilosort within spikeinterface.
I will read up on probeinterface docs. In the meantime @alejoe91 let me know if there's anyway I can give you access to my data or something else you would need from me so you an check at your end if you are seeing the same pattern. As far as I know, I have run this with datasets recorded at multiple sessions and the same patten of mismatch continues.

zm711, 2024-04-24:
> By Cluster info I meant spike_clusters.npy file that kilosort creates. Doesnt that have the information on what unit is assigned to each cluster

Yep. Sorry that was me using spikeinterface terminology. So in spikeinterface we call those the unit_ids. Kilosort decided to call those cluster_ids, but they are the same thing!

And google drive link works if you want it public otherwise we can share an email with you if you prefer to share privately. Let us know.

taningh86, 2024-04-25:
Got it! Thanks.
and, yeah I don't mind the data in public. I will share a G-drive link in few hours.

taningh86, 2024-04-25:
here's the link https://drive.google.com/drive/folders/1-j_UIW5fbm1iI3c9B3w-Ih3MH25hp5w-?usp=drive_link

taningh86, 2024-04-25:
Also, forgot to mention I used KS2.5. The files I shared went through CatGT. But in Spikeinterface I don't apply any bandpass filetr but run it through CMR (even though it ran through CAR in catGT). I don't think the additional CMR should effect the data.

zm711, 2024-04-26:
Hey @taningh86, thanks. Will try to check this out soon unless Alessio gets to it first.

taningh86, 2024-04-26:
No worries @zm711. Thanks a bunch to both you and @alejoe91 for getting to the bottom of the issue.

taningh86, 2024-07-30:
Hi. Just getting back to see if anyone had a chance to look at the dataset I shared with you and look at why some Phy and spikeinterface unit locations don't match between the two. Thanks!

alejoe91, 2024-07-31:
Hi @taningh86 

Unfortunately not yet. We have been quite busy with a release and August is a bit off for us.

We'll try to take a look, but the problem is that this is a very tricky one to debug. Can you show how the templates look in SpikeInterface? Do they look similar to the KS ones? I expect also the templates will be located differently. This could be due to some high levels of noise at the tip of the probe, which "confuses" SpikeInterface in finding the correct channels for unit 1. Could this be the reason? In this case, you should remove bad channels prior to sorting (or postprocessing)

cecidella, 2024-10-29:
Hello! Has there been any update on this? I'm also seeing the same mismatch on unit locations between phy/si (also in the order of ~2000 microns). I have run spikesorting separately (both KS4 and KS2) and I am seeing consistently the same issue. I have checked the channel map I'm using in KS and it is the correct one (NPx 1.0, phase3b1 map). I mostly use SI to make the unit summary plots, I normally curate manually in phy. Is there a way to save the channel map spikeinterface would use to see if something changes when running KS with this map? or to use a specific one in spikeinterface? If I export to phy then I see the same channels in the two. I am just confused as to why reading the outputs from KS gives such different results... if it is the channel map then I would still like to figure out which one is the correct one to use for the future! Thank you!