SpikeInterface/spikeinterface #2725: `plot_motion` does not accept `axes` or `ax` argument

JoeZiminski, 2024-04-17:
The `widgets.motion.plot_motion()` function asserts that `ax` and `axes` is `None` [here](https://github.com/SpikeInterface/spikeinterface/blob/fa57feeca6b403ae8595204f18313f038f7dc5ac/src/spikeinterface/widgets/motion.py#L76) but the [documentation](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.widgets.plot_motion) says these should be accepted inputs.

Is there a reason for this? Happy to open a PR to make the function support `ax` / `axes` arguments if not.

Also the latest docs have a number of references to `si.load_motion_info` and `si.plot_motion` but it appears these have since moved, also happy to open PR ASAP but leaving here as don't want to forget.

JoeZiminski, 2024-04-17:
Yes indeed this makes sense, it works well with `figure` argument. I think in this case just a message needs to be added to assert and docs updated. Will open a PR on this if all agree.

zm711, 2024-04-17:
As far as assert messaging I don't think anyone would complain about that :) But we can definitely wait for the others to weigh in.

samuelgarcia, 2024-04-29:
Hi
In si, some matplotlib plotting method accept ax or axes externally, which is convinient to construct summary figures for instance. When multiple axes it is more tedious to handle it so we put I put this assert at some places which was more easy. Unfortunatly the doc is generic about this backend_kwargs.

zm711, 2024-04-29:
So I think the fix would be like what @JoeZiminski said. Add better assert messaging so the user knows this is one function that can't accept external ax/axes. Good for you @samuelgarcia ?

samuelgarcia, 2024-04-29:
yes yes of course (or making the code accept axes)

JoeZiminski, 2024-07-18:
Fixed in #3068

JoeZiminski, 2024-07-18:
Well, kind of, it still says in the documentations it is accepted but this is hard to fix, the assert message is clear so now it is immediately clear what to do, so fixed-enough.

closed 2024-07-18 by JoeZiminski