SpikeInterface/spikeinterface #2726: AssertionError: Some channel in stream_id 1 do not have same ['sampling_rate', 'dtype', 'stream_id'] []

Kayv-cmb, 2024-04-17:
Hello SpikeInterface,

I've come across this error reading file with `read_openephys()`

`AssertionError: Some channel in stream_id 1 do not have same ['sampling_rate', 'dtype', 'stream_id'] []`

Can I get some help?

Kayv-cmb, 2024-04-17:
My Script 
```
from spikeinterface.preprocessing import common_reference,bandpass_filter
import os
import spikeinterface.full as si
import spikeinterface.extractors as se 
import spikeinterface.sorters as ss
from pathlib import Path

import numpy as np
import pandas as pd
job_kwargs = dict(n_jobs=40, chunk_size=30000, progress_bar=True)

# Extracting info for the path from the CSV, can be manually input
path_1 = '/nemo/lab/schaefera/working/dellacc/Recordings/240411/2024-04-11_14-45-47'
stream_1 = 1
block_1 = 1
recording_oe_1 = se.read_openephys(path_1 ,stream_id =str(stream_1),block_index=block_1,)
```

Full trace
```

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[17], [line 17](vscode-notebook-cell:?execution_count=17&line=17)
     [15](vscode-notebook-cell:?execution_count=17&line=15) stream_1 = 1
     [16](vscode-notebook-cell:?execution_count=17&line=16) block_1 = 1
---> [17](vscode-notebook-cell:?execution_count=17&line=17) recording_oe_1 = se.read_openephys(path_1 ,stream_id =str(stream_1),block_index=block_1,)
     [18](vscode-notebook-cell:?execution_count=17&line=18) print(recording_oe_1)
     [19](vscode-notebook-cell:?execution_count=17&line=19) bad_channel_ids, channel_labels = si.detect_bad_channels(recording_oe_1)

File [~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:335](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:335), in read_openephys(folder_path, **kwargs)
    [332](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:332)     recording = OpenEphysLegacyRecordingExtractor(folder_path, **kwargs)
    [333](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:333) else:
    [334](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:334)     # format = 'binary'
--> [335](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:335)     recording = OpenEphysBinaryRecordingExtractor(folder_path, **kwargs)
    [336](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:336) return recording

File [~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:147](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:147), in OpenEphysBinaryRecordingExtractor.__init__(self, folder_path, load_sync_channel, load_sync_timestamps, experiment_names, stream_id, stream_name, block_index, all_annotations)
    [135](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:135) def __init__(
    [136](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:136)     self,
    [137](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:137)     folder_path,
   (...)
    [144](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:144)     all_annotations=False,
    [145](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/spikeinterface/extractors/neoextractors/openephys.py:145) ):
[... cut ...] ite-packages/neo/rawio/baserawio.py:442) # also check that channel_id is unique inside a stream
    [443](https://vscode-remote+ssh-002dremote-002bvis-005fnode1.vscode-resource.vscode-cdn.net/camp/home/combadk/working/combadk/ScriptKC/NP_spikesorting/Cecilia_sorting/~/.conda/envs/CHIME_kayvan/lib/python3.9/site-packages/neo/rawio/baserawio.py:443) channel_ids = signal_channels[mask]['id']

AssertionError: Some channel in stream_id 1 do not have same ['sampling_rate', 'dtype', 'stream_id'] []
```

alejoe91, 2024-04-18:
Could you share with us the open ephys folder either as a zip file or on a google drive link?

Kayv-cmb, 2024-04-18:
Hi I am currently trying to compress my folder, but it is quite big. If it can help until the compression is finished I have the xml of the settings
[settings.xml.zip](https://github.com/SpikeInterface/spikeinterface/files/15023712/settings.xml.zip)

Kayv-cmb, 2024-04-18:
I can't share my compressed openephys folder (25G) either here or with google drive link? Do you have an alternative solution?

h-mayorquin, 2025-03-13:
Hey, @Kayv-cmb,

Are you still interested on solving this issue? I might have some time now but this has been a while, maybe we should close this if this is no longer relevant.

closed 2025-03-13 by Kayv-cmb