SpikeInterface/spikeinterface #2730: 'ls' and 'du' not recognized as an internal or external command

mtehrani-code, 2024-04-17:
In the November 2023 the tutorial, I'm running into an issue with the !ls-llh command in several places and the !du -sh command in jupyter notebook:

1)Saving and loading si objects:

Command: !ls -llh {base_folder}/preprocessed 
Error: 'ls' is not recognized as an internal or external command,
operable program or batch file.

2) Data compression:

Command:  !du -sh {base_folder}/preprocessed_compressed.zarr
                    !du -sh {base_folder}/preprocessed
Error: 'du' is not recognized as an internal or external command,
operable program or batch file.

3) Run sorter locally

Command: !ls -llh {base_folder}/results_KS4
Error: 'ls' is not recognized as an internal or external command,
operable program or batch file.

It looks like the above errors might have to do with the fact that I'm using windows and ls at least is a linux command? What would be the correct way to write these in windows?

Thanks,
M

alejoe91, 2024-04-17:
Yes these are Linux commands. You can just check the content of the folder with the explore files and check the size from the folder properties

mtehrani-code, 2024-04-17:
Got it. Thanks.

closed 2024-04-17 by mtehrani-code