SpikeInterface/spikeinterface #2731: Waveform extraction: AttributeError: 'KiloSortSortingExtractor' object has no attribute 'get_probegroup'

mtehrani-code, 2024-04-17:
Still working with the Cambridge NeuroTech, November 2023 tutorial.

In the waveform extraction step, I type:

we = si.create_sorting_analyzer(recording_saved, sorting, folder=base_folder / "waveforms_dense", 
                          sparse=False, overwrite=True, **job_kwargs)
print(we)

And I get this error:

AttributeError                            Traceback (most recent call last)
Cell In[109], line 1
----> 1 we = si.create_sorting_analyzer(recording_saved, sorting, folder=base_folder / "waveforms_dense", 
      2                           sparse=False, overwrite=True, **job_kwargs)
      3 print(we)

File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:121, in create_sorting_analyzer(sorting, recording, format, folder, sparse, sparsity, overwrite, **sparsity_kwargs)
    118 else:
    119     sparsity = None
--> 121 sorting_analyzer = SortingAnalyzer.create(sorting, recording, format=format, folder=folder, sparsity=sparsity)
    123 return sorting_analyzer

File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:221, in SortingAnalyzer.create(cls, sorting, recording, format, folder, sparsity)
    219 assert sorting.sampling_frequency == recording.sampling_frequency
    220 # check that multiple probes are non-overlapping
--> 221 all_probes = recording.get_probegroup().probes
    222 check_probe_do_not_overlap(all_probes)
    224 if format == "memory":

AttributeError: 'KiloSortSortingExtractor' object has no attribute 'get_probegroup'

I'm replacing "extract_waveforms" with "create_sorting_analyzer" as was suggested in the update.

Please help!

alejoe91, 2024-04-17:
The create_sorting_analyzer takes a sorting object first and then a recording, not the other way around!

closed 2024-04-17 by alejoe91

mtehrani-code, 2024-04-17:
Okay. When I swap those, this is what I get:

SortingAnalyzer: 49 channels - 120 units - 1 segments - memory - has recording
Loaded 0 extensions: 

No processing happens...Shouldn't there be an output file like a 'waveforms' folder somewhere?

alejoe91, 2024-04-17:
No, this only creates a sorting analyzer, that can be used to compute extensions. Note that the waveform extraction step has been deprecated