SpikeInterface/spikeinterface #2732: dendro demo broken: parallel write

bendichter, 2024-04-18:
```bash
[dendro] Running job: 1b0ca739.e31f3a60
[dendro] Processor: spikeinterface_pipeline_mountainsort5
[dendro] Assembling context
[dendro] Input: input
[dendro] Output: output
[dendro] Parameter: job_kwargs.n_jobs = 0.5
[dendro] Parameter: job_kwargs.chunk_duration = 1s
[dendro] Parameter: job_kwargs.progress_bar = False
[dendro] Parameter: recording_context.electrical_series_path = /acquisition/ElectricalSeriesAp
[dendro] Parameter: recording_context.lazy_read_input = True
[dendro] Parameter: recording_context.write_recording = False
[dendro] Parameter: recording_context.stub_test = True
[dendro] Parameter: recording_context.stub_test_duration_sec = 300
[dendro] Parameter: recording_context.stub_test_num_channels = -1
[dendro] Parameter: run_preprocessing = True
[dendro] Parameter: preprocessing_context.add_preprocessed_to_output_nwb = False
[dendro] Parameter: preprocessing_context.preprocessing_strategy = cmr
[dendro] Parameter: preprocessing_context.highpass_filter.freq_min = 300
[dendro] Parameter: preprocessing_context.highpass_filter.margin_ms = 5
[dendro] Parameter: preprocessing_context.phase_shift.margin_ms = 100
[dendro] Parameter: preprocessing_context.detect_bad_channels.method = coherence+psd
[dendro] Parameter: preprocessing_context.detect_bad_channels.dead_channel_threshold = -0.5
[dendro] Parameter: preprocessing_context.detect_bad_channels.noisy_channel_threshold = 1
[dendro] Parameter: preprocessing_context.detect_bad_channels.outside_channel_threshold = -0.3
[dendro] Parameter: preprocessing_context.detect_bad_channels.n_neighbors = 11
[dendro] Parameter: preprocessing_context.detect_bad_channels.seed = 0
[dendro] Parameter: preprocessing_context.common_reference.reference = global
[dendro] Parameter: preprocessing_context.common_reference.operator = median
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.n_channel_pad = 60
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.n_channel_taper = None
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.direction = y
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.apply_agc = True
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.agc_window_length_s = 0.01
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.highpass_butter_order = 3
[dendro] Parameter: preprocessing_context.highpass_spatial_filter.highpass_butter_wn = 0.01
[dendro] Parameter: preprocessing_context.motion_correction.strategy = skip
[dendro] Parameter: preprocessing_context.motion_correction.preset = nonrigid_accurate
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.detect_kwargs.method = locally_exclusive
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.detect_kwargs.peak_sign = neg
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.detect_kwargs.detect_threshold = 8
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.detect_kwargs.exclude_sweep_ms = 0.1
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.detect_kwargs.radius_um = 50
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.localize_peaks_kwargs.radius_um = 75
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.localize_peaks_kwargs.max_distance_um = 150
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.localize_peaks_kwargs.optimizer = minimize_with_log_penality
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.localize_peaks_kwargs.enforce_decrease = True
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.localize_peaks_kwargs.feature = ptp
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.method = decentralized
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.direction = y
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.bin_duration_s = 2
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.rigid = False
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.bin_um = 5
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.margin_um = 0
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.win_shape = gaussian
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.win_step_um = 100
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.win_sigma_um = 200
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.histogram_depth_smooth_um = 5
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.histogram_time_smooth_s = None
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.pairwise_displacement_method = conv
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.max_displacement_um = 100
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.weight_scale = linear
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.error_sigma = 0.2
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.conv_engine = numpy
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.torch_device = 
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.batch_size = 1
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.corr_threshold = 0
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.time_horizon_s = None
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.convergence_method = lsmr
[dendro] Parameter: preprocessing_context.motion_correction.motion_kwargs_nonrigid_accurate.estimate_motion_kwargs.soft_weights = False
[... 235 lines cut ...]
    return fut.result(timeout)
  File "/usr/local/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/usr/local/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.

Spike sorting failed. You can inspect the runtime trace in /tmp/working/scratch/tmp_spikesorting/spikeinterface_log.json.
Traceback (most recent call last):
  File "/app/main.py", line 44, in <module>
    app.run()
  File "/src/dendro/python/dendro/sdk/App.py", line 84, in run
    return _run_job_child_process(job_id=JOB_ID, job_private_key=JOB_PRIVATE_KEY, processors=self._processors)
  File "/src/dendro/python/dendro/sdk/_run_job_child_process.py", line 141, in _run_job_child_process
    processor_class.run(context)
  File "/app/main.py", line 37, in run
    run_pipeline(context)
  File "/app/common/processor_pipeline.py", line 116, in run_pipeline
    recording_preprocessed, sorting, waveform_extractor, sorting_curated, visualization_output = si_pipeline.run_pipeline(
  File "/src/spikeinterface_pipelines/src/spikeinterface_pipelines/pipeline.py", line 114, in run_pipeline
    raise Exception("Spike sorting failed")
Exception: Spike sorting failed
```

This occurred during @luiztauffer 's [Dendro tutorial](https://flatironinstitute.github.io/dendro-docs/docs/get_started)

alejoe91, 2024-04-18:
Thanks @bendichter 

This is a known issue and we pushed a patch on the MS5 wrapper to fix it. I guess @luiztauffer still needs to update the app image to include that fix

bendichter, 2024-04-18:
oh, good to know. Thanks!

zm711, 2024-06-19:
I close this for now, but let us know if it is still a problem.

closed 2024-06-19 by zm711