SpikeInterface/spikeinterface #2734: sorting analyzer has no extensions

mtehrani-code, 2024-04-18:
Hello,

I ran the data through the sorter (KS4) and I'm at the post-processing step.

I created the sorting_analyzer as follows:

sorting_analyzer = si.create_sorting_analyzer(sorting, recording_saved, folder=base_folder / "waveforms_dense", 
                          sparse=False, overwrite=True, **job_kwargs)
print(sorting_analyzer)

and this is what I get:

SortingAnalyzer: 49 channels - 120 units - 1 segments - memory - has recording
Loaded 0 extensions: 

Any idea why no extensions have been loaded? The following steps (i.e. sparsity, compute_principal_components, etc.) aren't working either likely as a result. Also, I skipped the waveform extraction step since it's been deprecated.

I did look through the documentation but not finding my answer.

Thanks,
M

zm711, 2024-04-18:
Howdy @mtehrani-code!

The `loaded 0 extensions` refers to things you've calculated. Since you haven't calculated anything yet there are no extensions!

So you need to do a couple things: but let's start with an easy one

```python
sorting_analyzer.compute('random_spikes')
print(sorting_analyzer)
```
Now you should see that you have 1 extension loaded `random_spikes`. Basically you need to follow a cascade to get this to work (explained in [this tutorial](https://spikeinterface.readthedocs.io/en/latest/how_to/waveform_extractor_to_sorting_analyzer.html) and [this doc](https://spikeinterface.readthedocs.io/en/latest/modules/core.html)). 

In short the sorting analyzer has a lot of parent-child relationships that can roughly be summarized as 
random_spikes=> waveforms=>templates 
noise_levels 

Once you have those four "core" extensions then you calculate everything else. So if I just want to use the defaults for everything I would do
```python
sorting_analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels',])
print(sorting_analyzer)
```

Note that your analyzer is only living in memory so to save your work you'll need to do `save_as` and choose either `zarr` or `binary_folder` backend. Give that a try and let us know!

mtehrani-code, 2024-04-18:
That worked. Thank you!

closed 2024-04-18 by mtehrani-code

mtehrani-code, 2024-04-19:
@zm711 quick question: I computed the extensions and have the sorting_analyzer saved as below:

sorting_analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels', 'spike_amplitudes'])
sorting_analyzer.save_as(folder='/Users/matta/Documents/SortingAnalyzer/Extensions', format="binary_folder")

Then I get this:
SortingAnalyzer: 49 channels - 120 units - 1 segments - memory - sparse - has recording
Loaded 5 extensions: random_spikes, noise_levels, waveforms, templates, spike_amplitudes

I must not be saving properly, because it is still listed as "memory" above. Also, because I'm getting errors later when I try to load or get extensions:

sorting_analyzer.get_saved_extension_names()
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[236], line 1
----> 1 sorting_analyzer.get_saved_extension_names()

File ~\Documents\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:952, in SortingAnalyzer.get_saved_extension_names(self)
    949                 saved_extension_names.append(extension_name)
    951 else:
--> 952     raise ValueError("SortingAnalyzer.get_saved_extension_names() works only with binary_folder and zarr")
    954 return saved_extension_names

And here:

sorting_analyzer.load_extension(extension_name="spike_amplitudes")
AssertionError: SortingAnalyzer.load_extension() does not work for format='memory' use SortingAnalyzer.get_extension() instead

What is the correct format to save the analyzer if what I have is incorrect?

Thanks!

M

zm711, 2024-04-19:
@mtehrani-code, yep no worry. It's a common mistake (I make it myself pretty often). There is a concept in some python libraries of `inplace` operations. That means changing your variable when you run code. (`save_as` for example). But SpikeInterface made the choice NOT to do `inplace` operations. So a variable isn't changed out from under you. So what you need to do instead is
```python
sorting_analyzer = sorting_analyzer.save_as(xx)
```

This is because the save_as returns a new version of a sorting_analyzer rather than modifying the current one in place. In face you could keep them both around

```python
sorting_analyzer_binary = sorting_analyzer.save_as()
```
But that is up to you!

mtehrani-code, 2024-04-19:
@zm711 got it. Thank you!