SpikeInterface/spikeinterface #2739: channel_ids in plot_probe_map function does not work

RobertoDF, 2024-04-22:
Hi,

The channel_ids argument in plot_probe_map (src/spikeinterface/widgets/probe_map.py) does not filter selected channels:

```
import spikeinterface as si  # import core only
import spikeinterface.extractors as se

local_path = si.download_dataset(remote_path="mearec/mearec_test_10s.h5")
recording, sorting_true = se.read_mearec(local_path)

fig, ax = plt.subplots(figsize=(10, 10))
si.plot_probe_map(recording, ax=ax, channel_ids=["1","2"], with_channel_ids=True, contacts_colors="r")
```

alejoe91, 2024-04-22:
Thanks @RobertoDF

While clearly the `channel_ids` argument doesn't do what it's supposed to do, I wonder whether it would be a useful feature to only plot some of the channels. Maybe instead we could have a `highlight_channel_ids` that colorse them differently with respect to the rest?

What do you think?

alejoe91, 2024-04-22:
@RobertoDF indeed! `contact_ids` are the physical contact name, channel IDs are the channel names given by the recording system. The important point is that the contacts are sorted according to the channel ids and this is why `text_on_contact` is build by slicing the channel ids

RobertoDF, 2024-04-22:
Sure! This seems to work indeed: https://github.com/RobertoDF/spikeinterface/blob/main/src/spikeinterface/widgets/probe_map.py 
I substituted the unused `channel_ids` with `color_channels`

PR #2740 (2024-05-13): Add `color_channels` option in in plot_probe_map
Fix #2739
closed 2024-05-13 by alejoe91