SpikeInterface/spikeinterface #2743: dumb question about custom probe desing.

labcnUC, 2024-04-22:
Hi, I'm currently analyzing some data, and I've created a custom probe comprising a series of planar electrodes arranged in a specific configuration. All of these tetrodes have their tips situated within the same plane (horizontal plane, lets callit "temporo-medial axis"), Consequently, I've assigned them a Z-coordinate value of 0 and distributed the electrodes across the X-Y plane.
However, upon observing graphs of neuropixels, I noticed that their contacts are situated within the same plane, but along the dorso-ventral axis. Furthermore, they also display graphs in the X-Y plane. Could I be making a mistake in the design of my probe? How can I create planar electrodes aligned with the horizontal plane?

This is sample of my code (in spanish):
`# Función para crear una sonda con 16 contactos circulares de 30 micrómetros
def create_probe():
    # Creamos una instancia de la clase Probe
    probe = Probe(ndim=2, si_units='um')

    positions = []
    for i in range(4):
        for j in range(4):
            x = i * 33 # Espaciamiento horizontal de 100 micrómetros
            y = j * 33  # Espaciamiento vertical de 100 micrómetros
            positions.append([x, y])

    # Configuramos las propiedades de los contactos
    shapes = ['circle'] * 16  # Todos los contactos son circulares
    shape_params = [{'radius': 15}] * 16  # Diámetro de 30 micrómetros

    # Establecemos los contactos en la sonda
    probe.set_contacts(positions=positions, shapes=shapes, shape_params=shape_params)

    return probe`

alejoe91, 2024-04-23:
Hola!

So just a premise: for spike sorting, the orientation of the electrodes with respect to the brain doesn't matter. It's only used to find adjacent electrodes.

Going back to your question: in our convention we use y as depth, so if you want to define your electrodes in the horizontal plane they should lie on the x-z axis

Cheers
Alessio

labcnUC, 2024-04-23:
Great, it’s a good point regarding spatially distributed probes.

Muchas gracias!

zm711, 2024-04-25:
Great let us know if you have more questions. I'll close this, but feel free to open new issues!

closed 2024-04-25 by zm711