SpikeInterface/spikeinterface #2745: Delete_extension on recompute

zm711, 2024-04-23:
So we were testing recomputing a full set of analysis. I wonder if this a windows error. Basically to recreate:

```python

analyzer = si.create_sorting_analyzer(sorting, rec_cmr, format='binary_folder', folder='path')
analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels', 'spike_amplitudes', 'spike_locations'])
analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels', 'spike_amplitudes', 'spike_locations'])
```

Error:

```python
ValueError: SortingAnalyzer need extension 'templates' to be computed
```

The extensions that are left behind based on the repr is

```
SortingAnalyzer: 128 channels - 4 units - 1 segments - binary_folder - sparse - has recording
Loaded 6 extensions: random_spikes, noise_levels, correlograms, spike_amplitudes, spike_locations, principal_components
```

I wonder if it the delete extension is missing some of the children because it is deleting in the wrong order? Or maybe because templates can depend on either random_spikes or waveforms the delete_extension on recompute is now not working as expected. Or of course It is a Windows error.....

full stack trace in case we need it

```python
  Cell In[36], line 1
    analyzer.compute('random_spikes')
  File ~\User\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:853 in compute
    return self.compute_one_extension(extension_name=input, save=save, **kwargs)
  File ~\UserH\Documents\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:909 in compute_one_extension
    self.delete_extension(child)
  File ~\User\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:1113 in delete_extension
    ext = self.load_extension(extension_name)
  File ~\Users\GitHub\spikeinterface\src\spikeinterface\core\sortinganalyzer.py:1090 in load_extension
    extension_instance = extension_class(self)
  File ~\User\GitHub\spikeinterface\src\spikeinterface\postprocessing\spike_locations.py:62 in __init__
    extremum_channel_inds = get_template_extremum_channel(self.sorting_analyzer, outputs="index")
  File ~\User\GitHub\spikeinterface\src\spikeinterface\core\template_tools.py:150 in get_template_extremum_channel
    peak_values = get_template_amplitudes(templates_or_sorting_analyzer, peak_sign=peak_sign, mode=mode)
  File ~\User\Documents\GitHub\spikeinterface\src\spikeinterface\core\template_tools.py:86 in get_template_amplitudes
    before = _get_nbefore(templates_or_sorting_analyzer)
  File ~\User\Documents\GitHub\spikeinterface\src\spikeinterface\core\template_tools.py:50 in _get_nbefore
    raise ValueError("SortingAnalyzer need extension 'templates' to be computed")
```

We are 11 commits behind main for this analysis, but there weren't any analyzer changes in those commits so this should be a current issue on Main.

chrishalcrow, 2024-04-24:
Hello, the following **doesn't** reproduce the error on my machines:

```
from spikeinterface.core import (
    generate_ground_truth_recording,
    create_sorting_analyzer,
)

recording, sorting = generate_ground_truth_recording(
    durations=[
        50.0,
    ],
    sampling_frequency=30_000.0,
    num_channels=6,
    num_units=10,
    generate_sorting_kwargs=dict(firing_rates=6.0, refractory_period_ms=4.0),
    noise_kwargs=dict(noise_levels=5.0, strategy="tile_pregenerated"),
    seed=2205,
)

analyzer = create_sorting_analyzer(sorting, recording, format="binary_folder", folder="a_sorting_analyzer_folder")

analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels', 'spike_amplitudes', 'spike_locations'])
analyzer.compute(['random_spikes', 'waveforms', 'templates', 'noise_levels', 'spike_amplitudes', 'spike_locations'])
```

Don't know if that's helpful...

zm711, 2024-04-24:
Which OS are you using Chris? Are you on linux or mac?

zm711, 2024-04-24:
One more question. When it failed for us it deleted the contents of the templates folder, but didn't delete the templates folder itself. Was that true during your testing as well?

chrishalcrow, 2024-04-24:
> One more question. When it failed for us it deleted the contents of the templates folder, but didn't delete the templates folder itself. Was that true during your testing as well?

Yes: on the failed runs I end up with an empty "templates" folder.

chrishalcrow, 2024-04-24:
Is it easy for you to try and run this a few times and see if it ever succeeds? Seems weird, but happens on my machine.

zm711, 2024-04-24:
I repeated what @chrishalcrow did and found that it took 3 tries to fail. No fancy stats, but there is a non-zero fail rate.

zm711, 2024-04-24:
I think I have an idea why this is happening!

So here we delete our extensions...
https://github.com/SpikeInterface/spikeinterface/blob/163ab282645e62fbfd75cd845eb7755da187c787/src/spikeinterface/core/sortinganalyzer.py#L965-L967

Which requires us to recursively search for all children dependencies here:
https://github.com/SpikeInterface/spikeinterface/blob/163ab282645e62fbfd75cd845eb7755da187c787/src/spikeinterface/core/sortinganalyzer.py#L1183-L1190

But notice that we take the `set` of these dependencies which scramble the order of the dependences. 

Then we need to delete:
https://github.com/SpikeInterface/spikeinterface/blob/163ab282645e62fbfd75cd845eb7755da187c787/src/spikeinterface/core/sortinganalyzer.py#L1105-L1117

But in order to delete we have to load the ext first. 

So what I think is happening is that doing `set` is adding some randomness into order in which we delete. So if we delete templates first, and then try to delete spike_locations then it fails because we can't load spike_locations. But as long as we delete templates after things that rely on templates then we are fine. If I add a print statement before delete we see

```
waveforms
amplitude_scalings
unit_locations
principal_components
templates
spike_amplitudes
template_metrics
quality_metrics
template_similarity
spike_locations
```

```
template_similarity
quality_metrics
spike_amplitudes
waveforms
template_metrics
amplitude_scalings
templates
principal_components
spike_locations

```

On the failures that templates was deleted before spike_locations.

And a success
```
unit_locations
template_metrics
spike_amplitudes
template_similarity
principal_components
spike_locations
templates
amplitude_scalings
waveforms
quality_metrics
unit_locations
template_metrics
spike_amplitudes
template_similarity
principal_components
spike_locations
templates
amplitude_scalings
quality_metrics
unit_locations
template_metrics
spike_amplitudes
template_similarity
spike_locations
amplitude_scalings
quality_metrics
```

We see that locations are deleted before templates!

PR #2757 (2024-04-29): Remove randomness of order in _get_children_dependencies
Fixes #2745 , bug discovered and solution discovered by @zm711 

The function `_get_children_dependencies` randomised the list of children, through the `set` function. This is a problem when using this function to delete extensions, since this has an ordering to it. Solution is simply to delete `set`, which didn't do anything (as far as I can tell, from playing with more complicated dependencies in the postprocessing module).

Added test to check that ordering is correct for the core extensions. Note: the test randomly passes or fails ra [... cut]
closed 2024-04-29 by alejoe91