SpikeInterface/spikeinterface #2754: Exported Phy data sets all clusters to "unsorted" group, ignoring sorter's group assignments

cheydrick, 2024-04-24:
Hi this is Chris with Plexon. I used SpikeInterface to sort using Kilosort 4, and then exported to Phy.

I found that when exporting to Phy that cluster is set to the “unsorted” group in cluster_group.tsv. The Kilosort 4 output also contains a cluster_group.tsv, but the clusters are set to either “mua” or “good”. 

The sorting object returned by run_sorter() contains this group assignment in the 'KSLabel' property. 

Is there a way for the Phy export process to carry over the cluster groups that Kilosort 4 assigned?

Thanks,
Chris
chris@plexon.com

zm711, 2024-04-25:
The `export_to_phy` doesn't automatically do that because not all sorters return `KSLabel`. Since that is a KS specific thing. I'm wondering if we can instead add a parameter to `export_to_phy` called `add_annotations` (or something like that) and then we can write any annotations that occur on a per-cluster basis into the tsv with a column heading of the key in the annotations dict. I don't use annotations for these objects much.... does that sound possible @alejoe91. Then if the user wants the KSLabel or any other label (for example they do qc metrics and make their own "score" of 4 of the metrics and want to create a column based on that for use in phy).

zm711, 2024-07-12:
Hey @cheydrick, 

This [PR](https://github.com/SpikeInterface/spikeinterface/pull/3123/files) allows you to export additional properties to your phy results so you can keep the KSLabel. You want to test out main and see if this works for you? 

Or if you've already tested this are you good to go?

cheydrick, 2024-07-13:
Thanks for giving me a heads up on the new functionality.

Adding "additional_properties=["KSLabel"]" as an argument to export_to_phy() included in the export a file called "cluster_KSLabel.tsv". Phy apparently knows to add this as a column to the cluster view.

[image]

I appreciate that you implemented this - hopefully it will be of use to others, too!

alejoe91, 2024-07-13:
Great! I'll close this then :)

closed 2024-07-13 by alejoe91