SpikeInterface/spikeinterface #2758: Question about the detection threshold

Antho2422, 2024-04-25:
Hi all, 

I have a question concerning the detection threshold. I am trying to add an horizontal line to my plot of detected spikes on traces. 

To get the y value of the line I do : threshold_value * noise_levels (that I got from the sorting analyzer extension) and this is the result I get : 

[image]

This is a noise cluster so do not mind the quality of the detection. The issue here is that some of the detected events seems to be below the line of threshold and I do not know exactly why ?

I remember that Pierre told me someday that there was a 0.67 factor on the traces and I would like to know if I should take it into account or if this can be coming from something else ? 

Thanks, 
Anthony

zm711, 2024-04-25:
@yger, seems like you discussed this. I haven't looked in the widgets for this so would you mind answering this one?

yger, 2024-04-25:
However, be careful because si.get_noise_levels() has a default mode with return_scaled=True. So if your data have a scaling, be cautious: detect_peaks internally deals with unscaled data. So do not detect noise levels yourself with return_scaled=True, and provide them to detect_peaks that way

Antho2422, 2024-04-25:
Thanks for the answer. I did not detect the peaks manually, this was the results of peaks detected using SC2. I just wanted to visually inspect what was detected on the raw traces using the appropriated plot function from SI and then plotting on top of that a line corresponding to the threshold + the noise level