SpikeInterface/spikeinterface #2764: add columns to unit table in plot_sorting_summary

RobertoDF, 2024-04-26:
Hi,
I don´t understand how to add some data to the unit_table showed in plot_sorting_summary. I see that the `unit_table_properties` arg should do that, but how to use it? The following doesn´t work:

```plot_sorting_summary(analyzer,` curation=True, backend='sortingview', unit_table_properties=["amplitude_cutoff"])```

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[58], line 1
----> 1 plot_sorting_summary(analyzer, curation=True, backend='sortingview', unit_table_properties=["amplitude_cutoff"])

File /alzheimer/Roberto/Software/mambaforge/envs/spikesorting/lib/python3.9/site-packages/spikeinterface/widgets/sorting_summary.py:83, in SortingSummaryWidget.__init__(self, sorting_analyzer, unit_ids, sparsity, max_amplitudes_per_unit, min_similarity_for_correlograms, curation, unit_table_properties, label_choices, backend, **backend_kwargs)
     70     unit_ids = sorting.get_unit_ids()
     72 plot_data = dict(
     73     sorting_analyzer=sorting_analyzer,
     74     unit_ids=unit_ids,
   (...)
     80     max_amplitudes_per_unit=max_amplitudes_per_unit,
     81 )
---> 83 BaseWidget.__init__(self, plot_data, backend=backend, **backend_kwargs)

File /alzheimer/Roberto/Software/mambaforge/envs/spikesorting/lib/python3.9/site-packages/spikeinterface/widgets/base.py:87, in BaseWidget.__init__(self, data_plot, backend, immediate_plot, **backend_kwargs)
     84 self.backend_kwargs = backend_kwargs_
     86 if immediate_plot:
---> 87     self.do_plot()

File /alzheimer/Roberto/Software/mambaforge/envs/spikesorting/lib/python3.9/site-packages/spikeinterface/widgets/base.py:108, in BaseWidget.do_plot(self)
    106 def do_plot(self):
    107     func = getattr(self, f"plot_{self.backend}")
--> 108     func(self.data_plot, **self.backend_kwargs)

File /alzheimer/Roberto/Software/mambaforge/envs/spikesorting/lib/python3.9/site-packages/spikeinterface/widgets/sorting_summary.py:153, in SortingSummaryWidget.plot_sortingview(self, data_plot, **backend_kwargs)
    148         similarity_scores.append(
    149             vv.UnitSimilarityScore(unit_id1=u1, unit_id2=u2, similarity=similarity[i1, i2].astype("float32"))
    150         )
    152 # unit ids
--> 153 v_units_table = generate_unit_table_view(
    154     dp.sorting_analyzer.sorting, dp.unit_table_properties, similarity_scores=similarity_scores
    155 )
    157 if dp.curation:
    158     v_curation = vv.SortingCuration2(label_choices=dp.label_choices)

File /alzheimer/Roberto/Software/mambaforge/envs/spikesorting/lib/python3.9/site-packages/spikeinterface/widgets/utils_sortingview.py:62, in generate_unit_table_view(sorting, unit_properties, similarity_scores)
     60 property_values = sorting.get_property(prop_name)
     61 # make dtype available
---> 62 val0 = np.array(property_values[0])
     63 if val0.dtype.kind in ("i", "u"):
     64     dtype = "int"

TypeError: 'NoneType' object is not subscriptable
```

Thanks!

alejoe91, 2024-05-09:
Hi @RobertoDF 

The quality metrics are computed as an extension, but the are not set as properties to the sorting objects. In order to add `amplitude_cutoff` (or any other QM) as a column, you first need to set it as a property:

```
qm = analyzer.get_extension("quality_metrics").get_data()
# set property
analyzer.sorting.set_property("amplitude_cutoff", qm["amplitude_cutoff"].values)

# plot
sw.plot_sorting_summary(analyzer,` curation=True, backend='sortingview', unit_table_properties=["amplitude_cutoff"])
```

Let me know if that works!

JoeZiminski, 2024-06-21:
Hey @RobertoDF did you have any luck with that code suggestion?

RobertoDF, 2024-06-21:
Didn't have time to try this yet. But if u tested it we can close the issue.

closed 2024-06-21 by alejoe91