SpikeInterface/spikeinterface #2770: load_sorting_analyzer not defined

mtehrani-code, 2024-04-26:
Hi,

Working with the November 2023 tutorial, I created and saved the SortingAnalyzer:

SortingAnalyzer: 49 channels - 120 units - 1 segments - binary_folder - has recording
Loaded 0 extensions: 

But when I try to load it, I get the error:

sorting_analyzer=load_sorting_analyzer(folder="sorting_analyzer")
NameError: name 'load_sorting_analyzer' is not defined

Also, I can still calculate the extensions, but when I run:

pc0 = pc.get_projections(unit_id=0)
print(f"PC scores shape: {pc0.shape}")
all_labels, all_pcs = pc.get_all_projections()
print(f"All PC scores shape: {all_pcs.shape}")

I get:

AttributeError: 'ComputePrincipalComponents' object has no attribute 'get_projections'

Would appreciate your help trying to figure out the errors...

M

zm711, 2024-04-26:
How did you import `load_sorting_analyzer`? 

You probably need to do

```python
sorting_analyzer = si.load_sorting_analyzer(folder='sorting_analyzer')
```

Also what is your script for getting the pcs?

mtehrani-code, 2024-04-26:
I imported spikeinterface at the beginning:

import spikeinterface.full as si

Isn't load_sorting_analyzer within the spikeinterface.full module? Didn't know I was supposed to import it separately.

This is the script for the pc:

pc = sorting_analyzer.compute(input="principal_components", n_components=3, mode="by_channel_local")

M

zm711, 2024-04-26:
You imported the module with the alias of `si`. So whenever you want to use any code you have to start that code with the alias `si`. We explain how this works [here](https://spikeinterface.readthedocs.io/en/latest/import.html) and here is a random article about namespaces in python [article](https://realpython.com/python-namespaces-scope/). Basically in python if you import something with an alias you have to use that alias. This helps prevent polluting the name space. :) For example I could also do
```python
import spikeinterface.full as the_coolest_package_ever
analyzer = the_coolest_package_ever.load_sorting_analyzer()
```

We probably changed the name of that function (sorry during some code updates probably). See [here](https://github.com/SpikeInterface/spikeinterface/blob/747fc0b5eb5d0f6d02c30a67de95f6ab2b57299a/src/spikeinterface/postprocessing/principal_component.py#L123).

Basically it should be
```python
pc = sorting_analyzer.compute(xx)
pc0 = pc.get_projections_one_unit(0)
```

mtehrani-code, 2024-04-26:
That all worked. Thank you!

Now when I want to look at data associated with the extensions using get_data( ), nothing is returned. Eg.:

unit_locations = sorting_analyzer.get_extension (extension_name="unit_locations")
unit_locations_data = unit_locations.get_data()

I computed (or think I did) unit_locations earlier:

unit_locations=sorting_analyzer.compute("unit_locations", method="monopolar_triangulation")

I tried this on a couple of other extensions (templates and waveforms) and I get the same thing, i.e. no output.

What am I missing?

mtehrani-code, 2024-04-26:
...I am able to plot the unit locations however.

zm711, 2024-04-26:
I just tried that on my computer and it worked fine. If you just type:

```
sorting_analyzer
```
into your console which extensions show up?

mtehrani-code, 2024-04-29:
When I type sorting_analyzer, I get:

SortingAnalyzer: 49 channels - 120 units - 1 segments - binary_folder - has recording
Loaded 6 extensions: random_spikes, waveforms, templates, noise_levels, principal_components, unit_locations

zm711, 2024-04-29:
What if you just type
```python
sorting_analyzer.get_extension('unit_locations')
```

mtehrani-code, 2024-04-29:
Then I get:

<spikeinterface.postprocessing.unit_localization.ComputeUnitLocations at 0x1bfb0b41e50>

zm711, 2024-04-29:
Well that didn't return None. Cool. So now do 

```python
unit_locations = sorting_analyzer.get_extension('unit_locations')
unit_locations.get_data()
```

mtehrani-code, 2024-04-29:
That worked! Got a long array of numbers.

Hmm, I tried this version earlier, but it didn't return anything:

unit_locations = sorting_analyzer.get_extension(extension_name="unit_locations")
unit_locations_data = unit_locations.get_data()

Anyway, thanks for your help!!

zm711, 2024-04-29:
I see what happened!

unit_locations = sorting_analyzer.get_extension (extension_name="unit_locations")

You have an extra space between extension and (. 

You need to type `sorting_analyzer.get_extension(extension_name="unit_locations")`.

We can close this, but as always just open a new issue if something comes over.

closed 2024-04-29 by zm711

mtehrani-code, 2024-04-29:
Ah, I see. Thank you!