SpikeInterface/spikeinterface #2782: sortingview attribute error

mtehrani-code, 2024-04-29:
Not having luck with any of the viewers at the moment :/

When I type:

w = si.plot_sorting_summary(curated_sorting, backend="sortingview", curation=True)

I get the error:

AttributeError: 'UnitsSelectionSorting' object has no attribute 'has_extension'

Not sure where the error is coming from. I installed sortingview and did the kachery authentication (I think).

Thanks in advance.

M

alejoe91, 2024-04-29:
Hi,

As mentioned in the docs, the function needs a SortingAnyzer, not a Sorting object.

zm711, 2024-04-30:
Maybe we should add a warning here: https://github.com/SpikeInterface/spikeinterface/blob/b6c2c91ecfc71dfa44213eaa30bf8dacf2da72a7/src/spikeinterface/widgets/base.py#L117-L118

Or maybe in the individual widgets themselves to make it clearer to the end-user when a sorting is acceptable and when you must have a sorting_analyzer. Although the argument names try to make this clear `sorting_analyzer` vs `sorting_analyzer_or_sorting` you wouldn't know that pattern if you 1) only use positional args or 2) you only use one plotting function so you don't know the convention.

alejoe91, 2024-04-30:
but yeah that final else whould be a `ValueError`!!!

mtehrani-code, 2024-04-30:
@alejoe91 Yes, that worked! Thank you.

I did check the link to the docs for sortingview that was in the tutorial (below). But it looks like it is no longer available:

https://spikeinterface.readthedocs.io/en/latest/module_widgets.html

closed 2024-05-02 by samuelgarcia