SpikeInterface/spikeinterface #2787: PC-based quality metrics nan value meaning

zm711, 2024-04-30:
Quick question that I thought I would ask before I try to track it down myself. Sometimes all the pc-based metrics return `nan` instead of a value (d-prime, isolation distance, etc). for one or a couple units. What does that actually mean? Since these come from the pcs that must mean that something is wrong with the pcs just for that neuron, but would that mean that there is a problem with some waveforms, the random spikes or the method of calculating the pcs. is this fixable or is it just a "bad" unit?

alejoe91, 2024-05-09:
Hi @zm711 

The NaN values for the PC-based metrics is triggered [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/qualitymetrics/pca_metrics.py#L216-L220), so basically the mahalanobis distance between the unit projections and other units cannot be calculated. IMO, since the other units' projections are the ones from the units that share at least a channel within their sparse channels, this could mean that the unit doesn't have any "neighbor" units, which might be a good thing!

zm711, 2024-05-09:
That's true that would be good. NaNs make masking a little tricky though. `qm['isolation_distance']> x` will be False for nan. But I guess it'll just take a more complicated expression. :)

closed 2024-05-09 by zm711