SpikeInterface/spikeinterface #2789: Should aggregate channels preserve the channel_ids?

h-mayorquin, 2024-04-30:
The following code fails:

```python
from spikeinterface.core import aggregate_channels

recording1 = generate_recording(num_channels=3, durations=[10], set_probe=False)  # To avoid location check
recording1 = recording1.rename_channels(new_channel_ids=["a", "b", "c"])
recording2 = generate_recording(num_channels=2, durations=[10], set_probe=False)  
recording2 = recording2.rename_channels(new_channel_ids=["d", "e"])

aggregated_recording = aggregate_channels([recording1, recording2])  
assert aggregated_recording.get_num_channels() == 5
assert list(aggregated_recording.get_channel_ids()) == ['a', 'b', 'c', 'd', 'e']

```

The `channel_ids` of the aggregated recording are:

```
[0, 1, 2, 3, 4]
```

I think that the `channel_ids` should be preserved? Thoughts? Unforseen complexities?

zm711, 2024-05-01:
Could there ever be a case where the channel_ids are ['a', 'b', 'c'] and ['a', 'b'] such that by keeping them you would mess up  the indexing post-aggregating? I'm not sure why this would happen (combining two recordings that shouldn't be combined...?) but that is the only thing I could think of.

alejoe91, 2024-05-01:
Yes, in that case we should rename channels. IMO, if the channel ids are unique, then we can preserve them

samuelgarcia, 2024-05-01:
And also the same dtype.

closed 2024-05-14 by alejoe91