SpikeInterface/spikeinterface #2804: Loading SpikeGLX survey mode recordings

JulesLebert, 2024-05-04:
Hello,
SpikeGLX has a Survey mode (https://vimeo.com/783581937) to map the activity of all the banks on a neuropixels probe. It would be very useful to load these session in spikeinterface to compute some metrics on the entire probe. I imagine one way would be to aggregate the channels of the different bank recordings into one recording as if they were recorded simultaneously.

Here is some code from Jennifer Colonell that loads such recording in Matlab: https://github.com/jenniferColonell/SpikeGLX_Datafile_Tools/blob/main/MATLAB/SGLX_readMeta.m, function svyBankTimes.

The main difference with a normal recording is the addition of the field `~svySBTT` in the metadata, indicating the start sample for each shank/bank. I am unsure what the two sample numbers for each bank correspond to. My guess is that the first sample is when the recording changes bank and the second is when the amplifiers have settled.

I am including a file with such survey mode recording. Note that there is nothing interesting in this recording as it's just a probe in the air with a survey of only 5s per bank.

[Onedrive link](https://liveuclac-my.sharepoint.com/:u:/g/personal/skgtjml_ucl_ac_uk/EdwX8oGnlRhAq1K5FuaHBjcB2QNVPrhrz3ruZDNWav6EUA?e=JdQdpb) to the compressed recording

alejoe91, 2024-05-06:
Thanks @JulesLebert 

Realistically, we won't have time to tackle this before mid June. Maybe you'd like to take a stab at it?

JulesLebert, 2024-05-07:
> Thanks @JulesLebert
> 
> Realistically, we won't have time to tackle this before mid June. Maybe you'd like to take a stab at it?

Unfortunately, I will not have time to do this before at least October/November this year. But I would be happy to try then!

aidapiccato, 2025-04-17:
Hi! I'd also like to use this for probe surveys of long NHP probes. Is there any update on an extractor with this functionality?

aidapiccato, 2025-05-03:
Hi! I've made some progress using Jennifer Colonell's tool to load in activity from each probe bank and create an array that is (total num. channels per probe x num. timesteps per bank). I'd like to apply preprocessing functions to this dataset to eventually extract and plot peaks across the probe. Is there a straightforward way to load in raw data formatted like this into a recording object that I can apply processing functions to? Let me know if you'd like me to share the code with you.

alejoe91, 2025-05-04:
Hi @aidapiccato 

That is great! I think that we could move the issue to python-neo and discuss there, since readers are implemented there. It would be great if you could share the code, or if you feel confident also.make a PR :)

Thanks!

aidapiccato, 2025-05-07:
Hi @alejoe91,

I made a [feature request](https://github.com/NeuralEnsemble/python-neo/issues/1700) in python-neo. Let me know if any additional context would be useful.

Thanks!

zm711, 2025-07-17:
We are still waiting for some sample code and more info on the Neo side. Feel free to respond over there with more information so we can work on this.