SpikeInterface/spikeinterface #2807: feature request: irc version 1

WeissShahaf, 2024-05-06:
Dear Devs,

please add a wrapper for ironclust running the irc *** instead of the irc2 **** commands. 
several labs in our institute use ironclust from the irc command and irc2 is buggy and doesn't actually allow manual curation.

thanks,
Shahaf

alejoe91, 2024-05-06:
Hi @WeissShahaf 

It might be as easy as #2809

Can you try to run from that PR and try:
```
sorting_irc1 = ss.run_sorter("ironclust", recording, ..., version=1)
```

WeissShahaf, 2024-05-07:
i'm afraid not: 
Cell In[17], line 1
    sorting_irc1 = ss.run_sorter("ironclust", recording=rec,output_folder=savefolder,remove_existing_folder=True, version=1)

  File G:\mambaroot\envs\spkiPR\lib\site-packages\spikeinterface\sorters\runsorter.py:175 in run_sorter
    return run_sorter_local(**common_kwargs)

  File G:\mambaroot\envs\spkiPR\lib\site-packages\spikeinterface\sorters\runsorter.py:223 in run_sorter_local
    SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)

  File G:\mambaroot\envs\spkiPR\lib\site-packages\spikeinterface\sorters\basesorter.py:180 in set_params_to_folder
    raise AttributeError("Bad parameters: " + str(bad_params))

AttributeError: Bad parameters: ['version']

alejoe91, 2024-05-07:
@WeissShahaf did you install from source using the PR branch?

Here's what you should do:
```
pip uninstall -y spikeinterface
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
git checkout -b alejoe91-irc1 main
git pull https://github.com/alejoe91/spikeinterface.git irc1
pip install -e .
```

alejoe91, 2024-05-09:
@WeissShahaf did you get the chance to try my suggestion?

WeissShahaf, 2024-05-09:
yes. i'm kinda struggling with my IDE. (spyder). it insists on using the main spikeinterface definition of the spikeinterface modules instead of the pre-release one.

WeissShahaf, 2024-05-09:
##i finally managed to make it run. 

#sorting_irc1 = ss.run_sorter("ironclust", recording, ..., version=1) did not work, but this did:
other_params=ss.get_default_sorter_params('ironclust')
other_params['version']=1

sorting_irc1 = ss.run_sorter("ironclust", recording=rec,output_folder=savefolder,remove_existing_folder=True,**other_params)

## however, i crashed:

Traceback (most recent call last):

  Cell In[12], line 1
    sorting_irc1 = ss.run_sorter("ironclust", recording=rec,output_folder=savefolder,remove_existing_folder=True,**other_params)

  File D:\GitHub\NewGit\spkpre_release\spikeinterface\src\spikeinterface\sorters\runsorter.py:175 in run_sorter
    return run_sorter_local(**common_kwargs)

  File D:\GitHub\NewGit\spkpre_release\spikeinterface\src\spikeinterface\sorters\runsorter.py:225 in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)

  File D:\GitHub\NewGit\spkpre_release\spikeinterface\src\spikeinterface\sorters\basesorter.py:293 in run_from_folder
    raise SpikeSortingError(

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "D:\GitHub\NewGit\spkpre_release\spikeinterface\src\spikeinterface\sorters\basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "D:\GitHub\NewGit\spkpre_release\spikeinterface\src\spikeinterface\sorters\external\ironclust.py", line 268, in _run_from_folder
    raise Exception("ironclust returned a non-zero exit code")
Exception: ironclust returned a non-zero exit code

Spike sorting failed. You can inspect the runtime trace in E:\afm16505\231213\231213_neuropixels\231212_test4_g0_imec0\SI_out_test/spikeinterface_log.json.

## the json file shows that matlab did run. but then there was an error:

"File 1/1 took 1627.9s (81114.4 MB, 49.8 MB/s, x2.6 realtime)",
        "----------------------------------------Index in position 1 exceeds array bounds.ERROR: MATLAB error Exit Status: 0x00000001"
    ],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"D:\\GitHub\\NewGit\\spkpre_release\\spikeinterface\\src\\spikeinterface\\sorters\\basesorter.py\", line 258, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"D:\\GitHub\\NewGit\\spkpre_release\\spikeinterface\\src\\spikeinterface\\sorters\\external\\ironclust.py\", line 268, in _run_from_folder\n    raise Exception(\"ironclust returned a non-zero exit code\")\nException: ironclust returned a non-zero exit code\n",
    "run_time": null
}

closed 2024-05-13 by samuelgarcia

alejoe91, 2024-05-13:
I see..unfortunately that's an IronClust error, so not sure how we can help!

PR #2809 (2024-05-13): Add parameter for irc1
Fixes #2807