SpikeInterface/spikeinterface #2814: Docker 'APIError: 500 Server Error' when running Kilosort on Windows

m-beau, 2024-05-07:
I am experiencing a similar error attempting to run Kilosort in a docker container on a windows machine. Surprisingly, running the same sorter in a docker image on Spikeinterface's 'toy dataset' works like a charm, it only fails when trying to run it on actual data, so the issue seems to be related to reading this specific dataset. However, spikeinterface itself instanciates the recording object correctly without any problem. Here are the details:

OS: Windows 10. spikeinterface version: 0.101.0. Python version: 3.11.9. And nvidia-smi works well. Kilosort2 on MATLAB also runs correctly on this machine.

This works:
```python
import spikeinterface.full as si
test_recording, _ = si.toy_example(
     duration=30,
     seed=0,
     num_channels=64,
     num_segments=1)
sorting = si.run_sorter(sorter_name='kilosort4',
     recording=test_recording,
     output_folder="kilosort4_test",
     docker_image=True)
```

This returns an error:

```python
spikeglx_folder = r"D:\data_temp"
raw_rec = si.read_spikeglx(spikeglx_folder, stream_name='imec0.ap', load_sync_channel=False) # recording is correctly instanciated
sorting = si.run_sorter(sorter_name='kilosort4',
     recording=raw_rec,
     output_folder="kilosort4_test1",
     docker_image=True) # crash
```

This is the error log:

```bash
HTTPError                                 Traceback (most recent call last)
File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\api\client.py:265, in APIClient._raise_for_status(self, response)
    264 try:
--> 265     response.raise_for_status()
    266 except requests.exceptions.HTTPError as e:

File ~\Anaconda3\envs\lussac\Lib\site-packages\requests\models.py:1021, in Response.raise_for_status(self)
   1020 if http_error_msg:
-> 1021     raise HTTPError(http_error_msg, response=self)

HTTPError: 500 Server Error: Internal Server Error for url: http+docker://localnpipe/v1.45/containers/create

The above exception was the direct cause of the following exception:

APIError                                  Traceback (most recent call last)
Cell In[9], line 1
----> 1 sorting = si.run_sorter(sorter_name='kilosort4',
      2     recording=raw_rec,
      3     output_folder="kilosort4_test1", docker_image=True)

File ~\Anaconda3\envs\lussac\Lib\site-packages\spikeinterface\sorters\runsorter.py:169, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    167         else:
    168             container_image = singularity_image
--> 169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
    175 return run_sorter_local(**common_kwargs)

File ~\Anaconda3\envs\lussac\Lib\site-packages\spikeinterface\sorters\runsorter.py:457, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, **sorter_params)
    454     py_user_base_folder.mkdir(parents=True, exist_ok=True)
    455     py_user_base_unix = path_to_unix(py_user_base_folder)
--> 457 container_client = ContainerClient(mode, container_image, volumes, py_user_base_unix, extra_kwargs)
    458 if verbose:
    459     print("Starting container")

File ~\Anaconda3\envs\lussac\Lib\site-packages\spikeinterface\sorters\container_tools.py:97, in ContainerClient.__init__(self, mode, container_image, volumes, py_user_base, extra_kwargs)
     94         print(f"Docker: pulling image {container_image}")
     95         client.images.pull(container_image)
---> 97     self.docker_container = client.containers.create(container_image, tty=True, volumes=volumes, **extra_kwargs)
     99 elif mode == "singularity":
    100     assert self.py_user_base, "py_user_base folder must be set in singularity mode"

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\models\containers.py:932, in ContainerCollection.create(self, image, command, **kwargs)
    930 kwargs['version'] = self.client.api._version
    931 create_kwargs = _create_container_args(kwargs)
--> 932 resp = self.client.api.create_container(**create_kwargs)
    933 return self.get(resp['Id'])

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\api\container.py:439, in ContainerApiMixin.create_container(self, image, command, hostname, user, detach, stdin_open, tty, ports, environment, volumes, network_disabled, name, entrypoint, working_dir, domainname, host_config, mac_address, labels, stop_signal, networking_config, healthcheck, stop_timeout, runtime, use_config_proxy, platform)
    427     environment = self._proxy_configs.inject_proxy_environment(
    428         environment
    429     ) or None
    431 config = self.create_container_config(
    432     image, command, hostname, user, detach, stdin_open, tty,
    433     ports, environment, volumes,
   (...)
    437     stop_timeout, runtime
    438 )
--> 439 return self.create_container_from_config(config, name, platform)

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\api\container.py:456, in ContainerApiMixin.create_container_from_config(self, config, name, platform)
    454     params['platform'] = platform
    455 res = self._post_json(u, data=config, params=params)
--> 456 return self._result(res, True)

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\api\client.py:271, in APIClient._result(self, response, json, binary)
    269 def _result(self, response, json=False, binary=False):
    270     assert not (json and binary)
--> 271     self._raise_for_status(response)
    273     if json:
    274         return response.json()

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\api\client.py:267, in APIClient._raise_for_status(self, response)
    265     response.raise_for_status()
    266 except requests.exceptions.HTTPError as e:
--> 267     raise create_api_error_from_http_exception(e) from e

File ~\Anaconda3\envs\lussac\Lib\site-packages\docker\errors.py:39, in create_api_error_from_http_exception(e)
     37     else:
     38         cls = NotFound
---> 39 raise cls(e, response=response, explanation=explanation) from e

APIError: 500 Server Error for http+docker://localnpipe/v1.45/containers/create: Internal Server Error ("invalid volume specification: '/run/desktop/mnt/host/d:/:ro': invalid mount config for type "bind": invalid specification: destination can't be '/'")
```

Thank you for your help!

m-beau, 2024-05-07:
Further details: I am running spikeinterface from an interactive python shell, and the current working directory is located on the drive where the data sits AND where the output folder points:

[image]

alejoe91, 2024-05-07:
Can you try to install from source and add this print line [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/runsorter.py#L401)?

```
print(volumes, )
```

m-beau, 2024-05-07:
Volumes l.401 of spikeinterface/sorters/runsorter.py: {'D:\\': {'bind': '/', 'mode': 'ro'}, 'D:\\data_temp': {'bind': '/data_temp', 'mode': 'rw'}}

alejoe91, 2024-05-07:
Yeah, the first entry is problematic, but I'm not sure why it appears there..

Can you extend the print with:
```
print(volumes, recording_input_folders)
```

m-beau, 2024-05-07:
Volumes l.401 of spikeinterface/sorters/runsorter.py: {'D:\\': {'bind': '/', 'mode': 'ro'}, 'D:\\data_temp': {'bind': '/data_temp', 'mode': 'rw'}}
Volumes l.402 of spikeinterface/sorters/runsorter.py: [WindowsPath('D:/')]

alejoe91, 2024-05-07:
Ahhh I see what's happening! 

Since the SpikeGLX reader needs a folder path (`D:\\data_temp`), the parent folder is the entire `D:\` drive.

Can you try to make a child folder, e.g., `D:\\data_temp\spikeglx` and move the content of `data_temp` there? In this case, the parent folder is `d:\\data_temp`, which should be ok!

m-beau, 2024-05-07:
Aaah damn that was it. Thank you for your help, what an edge case!! Probably worth adding an assert statement when running a sorter on a spikeGLX recording to help future users :)

closed 2024-05-07 by m-beau

alejoe91, 2024-05-07:
Yeah I don't think it's related to SpikeGLX per se, but we should definitely add an assertion when the volume bindings are extracted!