SpikeInterface/spikeinterface #2816: Splitting units: What is the structure of indices_list?

emwonghiu, 2024-05-07:
Hi there,

Could you please explain what the structure of indices_list should look like?

"A list of index arrays selecting the spikes to split in each segment. Each array can contain more than 2 indices (e.g. for splitting in 3 or more units) and it should be the same length as the spike train (for each segment)"

From the docs, my understanding is indices_list is a list of arrays, and these arrays should have the same length as the spike train, so essentially it should be an array with a value for every spike, assigning each spike to one group or the other. 

For example, my data has 1 segment, and I want to split a unit based on amplitude, so I'll code the amplitudes into 2 indices, such that my indices_list looks like: [np.array([0, 0, 0, 1, 1, 0, ...])].

That is my understanding, but when I implement this in CurationSorting.split(), my sorting object becomes "None". 

Any help would be appreciated, thank you,
Emily

alejoe91, 2024-05-09:
Hi @emwonghiu 

As part of a PR #2775, we tried to improve the docstring of the split functions.

Can you see if the new [docstring](https://github.com/SpikeInterface/spikeinterface/blob/84219a5143cd28336d4ac5bf487b99e12d528f04/src/spikeinterface/curation/curationsorting.py#L65-L79) is clearer to you?

emwonghiu, 2024-05-09:
Hi @alejoe91 ,

Thank you, yes that is clearer. I also noticed that there's a new parameter, new_unit_ids. 

I have been using version 0.101.0. How do I get this version that you linked? Is it 0.100.6? (I'm new to git, have not tried anything besides pip install git+https://github.com/SpikeInterface/spikeinterface.git)

Thank you,
Emily

emwonghiu, 2024-05-09:
Update: I was able to checkout the pull request 2775, it's up and running.

However, I still get a NoneType object after applying CurationSorting().split(). In the source code I don't see any reason why a NoneType object would be produced. I've attached my code, maybe I input a wrong argument. 

[SplitUnit.txt](https://github.com/SpikeInterface/spikeinterface/files/15265156/SplitUnit.txt)

alejoe91, 2024-05-09:
The `split` function doesn't return anything, but it splits the `curation_sortomg.sorting`!

emwonghiu, 2024-05-09:
Oh my bad... I thought split() would return a new sorting object!

 I went back to the [tutorial](https://spikeinterface.readthedocs.io/en/latest/modules/curation.html?highlight=cs%20%3D%20CurationSorting#manual-curation) for this and followed the syntax:

`cs = scur.CurationSorting(parent_sorting=sorting_curated)
cs.split(
    split_unit_id=3, indices_list=np.array(coded_ampl), new_unit_ids=[50,51])
clean_sorting = cs.sorting

print("Clean sorting?", clean_sorting.get_unit_ids())  #[50, 51]

print(len(clean_sorting.get_unit_spike_train(unit_id=50)))  #0
print(len(clean_sorting.get_unit_spike_train(unit_id=51)))  #0

print(coded_ampl.count(50))  #64
print(coded_ampl.count(51))  #174
`

So now it produces the correct unit IDs (one unit split into 2: 50 and 51). But how do I get the spike trains? Where I'm getting 0's for the lengths of the spike trains, I should be getting 64 and 174, respectively.

Thanks for your help.

alejoe91, 2024-05-09:
Yeah you should get the same number. What is the content of `coded_ampl`?

alejoe91, 2024-05-10:
@emwonghiu there was indeed a problem, because the split expected the indices to be between 0 and n_splits-1.

This has been fixed by this PR #2775 which is now in main. Can you test again?

emwonghiu, 2024-05-10:
Once I changed coded_ampl to 0's and 1's, the splitting worked! I thought indices_list could contain any integers as long as they matched the unit ids given, but this makes more sense.

Thanks so much for your help!

closed 2024-05-10 by emwonghiu