SpikeInterface/spikeinterface #2817: export_to_phy() not respecting n_jobs

cheydrick, 2024-05-07:
The "Fitting PCA" step in the export to Phy process isn't respecting the n_jobs argument anymore.

I have set the global n_jobs parameter to 4 with:
si.set_global_job_kwargs(n_jobs=4)

When using SpikeInterface from April 17 (commit hash 33d478ab9) I can see that only four cores are being used while export_to_phy() is running.
[image]

However, when using SpikeInterface from May 6 (commit hash 42e1f024a) the "Fitting PCA" step is saturating all cores:
[image]

Thanks,
Chris
chris@plexon.com

zm711, 2024-05-07:
Maybe related to #2696... 

Thanks for the report. I'll see if I can track this down.

cheydrick, 2024-05-07:
@zm711 it's an int.

si.set_global_job_kwargs(n_jobs=4)

zm711, 2024-05-07:
Could you do me a favor and run with `verbose=True` inside the export_to_phy. There has been some work on `verbose` so it might not work but it would be nice to see the max amount of information. Specifically to see how many jobs are being passed to the `ChunkRecordingExecutor`.

zm711, 2024-05-07:
The other thing we may need to look at is this line where we send the `sorting.to_multiprocessing`. It seems like `NumpySorting` would be converted to `SharedMem` here. Is that desired? 
https://github.com/SpikeInterface/spikeinterface/blob/42e1f024aaa0c2af823f715a78d1d95a33807732/src/spikeinterface/postprocessing/principal_component.py#L351-L360

https://github.com/SpikeInterface/spikeinterface/blob/42e1f024aaa0c2af823f715a78d1d95a33807732/src/spikeinterface/core/basesorting.py#L651-L681

cheydrick, 2024-05-08:
@zm711 Looks like verbose is True by default.

https://github.com/SpikeInterface/spikeinterface/blob/42e1f024aaa0c2af823f715a78d1d95a33807732/src/spikeinterface/exporters/to_phy.py#L37

I explicitly set it to True anyways, but the output was the same as in my screenshots above.

zm711, 2024-05-08:
Unfortunately (and this is why verbose needs to be fixed), is that that verbose is just for the function and not at a global level passed into the other functions that accept verbose. So that's why it doesn't do what I wanted. I'll try to dig a bit more by running the PCA by itself and see if I can see what is going on.

zm711, 2024-05-08:
I think we have to wait for @samuelgarcia to comment on this one. But based on my reading the n_jobs being fed into the `ProcessPoolExecutor` is just a max number of processes. So technically the OS can move those processes around however it wants (for example we could also set n_jobs > number of cores and it would just have to schedule those extra processes onto the processors). This doesn't explain why the switch occurred with changing to NumpySorting from SharedMemSorting, but n_jobs cannot absolutely be guaranteed to be equal to the cores since the OS has a say in this.

alejoe91, 2024-07-02:
Looking at this again, I have no clue why this is maxing out the CPUs as `n_jobs` seems to be correctly propagated to the `ProcessPoolExecutor` [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/postprocessing/principal_component.py#L398)...

zm711, 2024-07-02:
I don't know why it would change with one of our commits, but n_jobs are just the number of processes not the number of cores so it could be a Windows update as well that changed how they were scheduling processes and so it was bouncing the processes to all the cores rather than limiting to specific cores. Seems inefficient, but not impossible right?

cheydrick, 2024-07-02:
@alejoe91 and @zm711 thanks for commenting on this - it made me take a look again, and I figured out the problem.

I decided to verify that commit 33d478a worked correctly with a fresh conda environment, and found that it also had the problem of saturating my processors.

Fortunately I had my old "good" conda environment from April 17 that didn't have this problem, and I compared the versions all of the installed packages between this "good" environment and the fresh environment. Long story short, changing scipy from version 1.13.1 to 1.12.0 fixed the problem. The "Fitting PCA" step no longer saturates my processor.

zm711, 2024-07-02:
Do you think we should add a note somewhere that scipy can saturate cores? So people don't blame us-- @alejoe91 ?

alejoe91, 2024-07-02:
@cheydrick Thanks for tracking this down!! Yeah let's keep this open so that we can investigate more

samuelgarcia, 2024-07-03:
My guess is the folowing:
n_jobs is the number of process and I guess this is correct.

In job_tools.py the number of thread per process is clip to 1 with max_threads_per_process and using the threadpoolctl package and so for save(n_jobs=4) you would not have the same. Is it right ?

But the pca fitting code (of Alessio) use ProcessPoolExecutor directly and is bypassing this max_threads_per_process and so scipy use as much as core as possible per process which is not efficient.

I guess that is this on the back of Alessio. Good luck camarade :)

alejoe91, 2024-07-03:
Good guess! So I should just add the max_threads_per_process

zm711, 2024-07-19:
I'm going to change the title of this issue to make clear that this is a ProcessPoolExcector issue in PCA.

zm711, 2024-12-03:
I think we should close this. The issue is in versions of scipy. We are working on trying to tweak our multiprocessing so maybe that will fix this in the future, but the solution would be to limit scipy if core number really needs to be respected.

closed 2024-12-03 by zm711