SpikeInterface/spikeinterface #2818: Optimizing spikeinterface speed

m-beau, 2024-05-07:
Quick question: Kilosort 4 in a spikeinterface docker container took about 4hours to run on a 20min long Neuropixels dataset (Kilosort 2 on MATLAB on the same machine takes about 30 minutes), so I believe that my installation is sub-optimal.

Is running sorters in containers knows to slow down the process, or am I doing something wrong? Do I need to check whether 'all my my GPU capabilities are used' somehow (not sure if it could fall back on running on CPU, but this would take much longer than 4h so I don't believe this must be happening)? In general, any tips to accelerate spikeinterface's sorters?

And also, any way to print out a progress bar of some kind? That would be very useful.

Thanks!

zm711, 2024-05-07:
I will say maybe (and we discussed this previously) we should have an option to write a binary and load that into KS instead of using their RecordingAsArray or whatever because that doesn't have multiprocessing so that might slowdown the process a bit. So in the case of someone sorting over the network it might make sense to just write the binary file where-ever they want and then sort from that for KS4 so that we can leverage our `n_jobs`, no?

zm711, 2024-05-08:
That sounds like a fun PR if you want to give it a go! I'm sure we would have users that would be interested in that :)

zm711, 2024-12-05:
I'll close this for now, but feel free to open more issues if other questions come up or if you decide to open a PR.

closed 2024-12-05 by zm711