SpikeInterface/spikeinterface #2821: Unable to find an appropriate device to wire to

ebach23, 2024-05-09:
We recently switched from a Cambridge Neurotech 64 channel probe to a Cambridge Neurotech 16 channel probe that uses an omnetics adapter to connect to our Intan 16 channel headstage. We are trying to analyze a group of our data obtained with this set-up but are not able to wire to any of the devices that are listed as an option for these probes. 
Our probes are: ASSY-1-P-2 - cambridgeneurotech - 16ch - 1shanks
The device options made available are:
['H32>RHD2132',
 'ASSY-156>RHD2164',
 'ASSY-116>RHD2132',
 'ASSY-77>Adpt.A64-Om32_2x-sm-NN>RHD2164',
 'ASSY-77>Adpt.A64-Om32_2x-sm-NN>two_RHD2132',
 'cambridgeneurotech_mini-amp-64']

All of these appear to be 32 to channel or 64 channel which then read the error: "AssertionError: chan_indices 32 != contact count 16"
Can you help us identify how we can fix this problem?

ebach23, 2024-05-09:
Thanks for your prompt response. I have posted this on the probeinterface repo, but may reach out to you about doing this manually. Is it best to reach you through this platform?

zm711, 2024-05-09:
I'm fine with here. Technically it is better to have the issue on the repo directly so have the issue on probeinterface, but that repo gets checked a little bit less. So we can work on this one here and then in the future open probe issues there and spikesorting stuff here :)

ebach23, 2024-05-14:
I got a response about this issue on the probe interface repo section, and gave them my reply. I havn't heard anything back. They had asked for the probe/adaptor map. Can you help me?

zm711, 2024-05-14:
Yep we can try to work on it. We are actually prepping for a conference/hackathon in a couple days so we are a little slower with replying (apologies!)

ebach23, 2024-05-29:
Just checking back in. I left the information on the [probeinterface repo] page in reply to your request (?). I havn't heard anything back. Anything I can help with on my end. Thanks!

zm711, 2024-05-31:
Hey @ebach23 thanks for pinging us. We are so sorry for being slow. We are at a developer hackathon for the project so we are all working on projects together. Could you ping us on Monday when the conference is over and someone will help you then? Thanks for understanding!

ebach23, 2024-06-03:
Hi zm711...per request...pinging you

zm711, 2024-06-03:
So is your question that you're not sure how to make your own mapping? Because you have all the information in order to make a mapping. My advice would be make the mapping and then submitting your mapping (on probeinterface) or here and then we can proofread and doublecheck for you.

If you don't know how to make a mapping we have a tutorial [here](https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24/tree/main/hands_on/probe_handling)

zm711, 2024-06-03:
It's a little tricky to explain over an issue text but the tutorial gives you a better idea of what goes into the mapping and it generates pictures as you run it. So I hope that is a better way to explain it! :)

ebach23, 2024-08-29:
Circling back around to data obtained using the probe/headstage combo I mention above. I used the tutorial to generate the manual mapping, but remain a bit confused and wanted to ensure that my logic wasn't incorrect. The 16 channel headstage 'use a 32-channel RHD amplifier chip. Amplifier channels 0-7 and 24-31 are tied to ground'. I initially tried to just use the numbers specified on the headstage which left me with a wiring of #manual_mapping=[22, 18, 17, 21, 23, 19, 16, 20, 12, 13, 11,9 ,10,  8, 14, 15]. Since in addition to the 16 channels we also have 8 ADC outputs this wiring did not throw me an error, but tried to plot 8 electrode channels and the 8 ADC outputs. I then subtracted the 8 from all the values of the headstage to arrive at 'manual_mapping=[14, 10, 9, 13, 15, 11, 8, 12, 4, 5, 3,1 ,2,  0, 6, 7]
probe.set_device_channel_indices(manual_mapping)"
Is that logic sound or am I missing something that will now cause incorrect wiring? Along that note, am I correct in thinking that our electrode adapter does not impact the wiring map of the probe/headstage combo and keeps omnetics outputs/headstage inputs aligned? Sorry for the novice questions! TIA for the help really appreciate you making this available :)