SpikeInterface/spikeinterface #2825: Is there a way to read/convert a .sev file?

victornovakov, 2024-05-09:
Hi, I apologize if this is an ignorant question, or has a quick solution. I have not been able to find it. 

I have raw electrode channel data in multiple .sev files which I would like to read into kilosort. Is there anyway to read .sev files or is it unsupported. I do not see it listed under supported file types and followed the direction to raise an issue. 

I also have the .sev files converted into a single .mat matlab file if this helps. 

Thank you

h-mayorquin, 2024-05-10:
For this type of question you should look into the extractors that we support. Check if the following solves supports your files:
 
 https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.extractors.read_tdt

h-mayorquin, 2024-05-23:
User has mentioned that the current solution in extractors is good enough. Closing this.

closed 2024-05-23 by h-mayorquin