SpikeInterface/spikeinterface #2831: Add `is_scaled` to `Templates` object (and zarr representatio)

alejoe91, 2024-05-10:
@h-mayorquin I think we should add a property to the `Templates` object to tell whether the `templates_array` is scaled to uV or not. 

This will be extremely important for hybrid generation. Currently, we are "assuming" that templates in the database are scaled, so my suggestion is to be explicit about it.

Any thoughts?

h-mayorquin, 2024-05-12:
I think this makes sense.  Is there a situation where we can to deal with raw templates? I am having a hard time imaging this. 

But I think it makes sense. Either that or we add a units property (.e.g "uV", "V" or "raw").

samuelgarcia, 2024-05-13:
Units is a good idea but if you go there this should be everywhere.
Now scaled = uV in all places.
So lets keep this for now and lets add units everywhere but later.

h-mayorquin, 2024-05-13:
This makes sense, Sam. I will do it like this.

zm711, 2024-05-14:
Closed by #2842.

closed 2024-05-14 by zm711