SpikeInterface/spikeinterface #2836: Python deprecation cycle

h-mayorquin, 2024-05-12:
Python 3.8 is coming to its EOL in September.

https://devguide.python.org/versions/

However, I want to make you aware that numpy follows a different cycle and within that python 3.8 already got deprecated

https://numpy.org/neps/nep-0029-deprecation_policy.html#support-table

I wanted to discuss how to handle this. I think given the centrality for this package and neo we should probably follow numpy cycle instead of the longer python one.

What do you think?

zm711, 2024-05-12:
Neo supports older numpy versions than the actual numpy EOL cycle. (It still tests 3.8 against 1.20.x). So if we go based on neo then it would basically still just be following the python release cycle.

h-mayorquin, 2024-05-12:
Numpy 1.20 has been deprecated for a year:
https://numpy.org/neps/nep-0029-deprecation_policy.html

Maybe something to discuss at neo. Any reason you guys are testing for that?

zm711, 2024-05-13:
I think the idea was that numpy does end of life pretty quick so as long as tests pass with it then it can be supported for longer (ie until the python version goes end of life).

JoeZiminski, 2024-05-15:
We follow the Numpy deprecation cycle for the same reason. I think it makes sense as it is a core module. Another benefit is that it is very easy to forget about python deprecation cycles, and so at least if there is a few months delay before you remember, you are still not behind the actual python deprecation.

h-mayorquin, 2024-05-16:
As a data point that presents evidence against my preference (following numpy deprecation cycle) today someone asked me if they could run the the library in python 3.7 (already deprecated for a while!) because of gpu issues (old nvida gpu I guess).

zm711, 2024-07-19:
Neo's deprecation cycle is set. It also follows quantities deprecation cycle. So it just depends on whether we want si to be stricter. But I tend to say if someone can run the code let them. We just don't test against it.