SpikeInterface/spikeinterface #2838: Remove spikes from spiketrains

Antho2422, 2024-05-13:
Hi there, 

I am trying to analyze data that I did not sorted myself. I just have the recording and Phy folder outputs to get the sortings. 

I also have "bad_periods" being tuples (start_sample, end_sample) of periods that I want to extract from the spike trains. The sorting has been done with those periods so I cannot use the silence_periods() function on the recording. 

My question is the following: What is the most efficient way to remove the spikes falling in those "bad_periods" ? Is there a function to do that and if not, how should I proceed ? 

Thanks, 

Anthony

zm711, 2024-06-21:
Sorry I think we missed this one Anthony.

I don't think we have a function? @alejoe91 @samuelgarcia ? You could probably just use the spike vector no? and make a function that pulls out the "bad spikes" and then load those spikes into a new sorting object? But maybe there is a better way?

Antho2422, 2024-07-02:
@zm711 

Hey, no problem !

I though about doing that but I was wondering if there was an easier way of doing that maybe ? But if not, I will make this work

Thanks,
Anthony

alejoe91, 2024-07-02:
Hi @Antho2422 

Nothing built in as @zm711 mentioned but it's pretty easy if you use the spike_vector representation:

```
from spikeinterface import NumpySorting

spikes_new = sorting.to_spike_vector().copy()

# this assumes you have one segment
for bad_period in bad_periods:
    spikes_index_to_remove_start, spikes_index_to_remove_stop = np.searchsorted(spikes_new["sample_index"], bad_period)
    spikes_new = np.delete(spikes_new np.arange(spikes_index_to_remove_start, spikes_index_to_remove_stop))
new_sorting = NumpySorting(spikes_new, unit_ids=sorting.unit_ids, sampling_frequency=sorting.sampling_frequency)
```

closed 2024-07-02 by Antho2422