SpikeInterface/spikeinterface #2844: ValueError in si.export_report and in si.plot_sorting_summary

meszena, 2024-05-14:
Hi! I would have a question:
I loaded back one of my external KS and Phy curated outputs to SortingAnalyzer and I've been able to calculate the various nice quality metrics but somehow I am not able to save the data (and the generated plots) as a report because (I think) there are NaNs or INFs in my unit metrics table (this is just a guess, maybe it is due to something else). 

Interestingly, I am still able to save the results as a .csv, like this:

_from spikeinterface.qualitymetrics import compute_quality_metrics
df = compute_quality_metrics(analyzer_clean)
df.to_csv(".../analyzer_Test_KS1/test.csv")_

 I just cannot do the "report" and the "web-based viewer".

Thanks so much in advance for any input!

I got the following warning and error when exporting the results:
---------------------------------------------------------------------------
**Warning:**
---------------------------------------------------------------------------
[C:\Users\cashlab\Documents\Domi\Python\spikeinterface\src\spikeinterface\widgets\unit_depths.py:77](file:///C:/Users/cashlab/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_depths.py:77): UserWarning: Attempting to set identical low and high xlims makes transformation singular; automatically expanding.
  ax.set_xlim(0, max(dp.unit_amplitudes) * 1.2)

---------------------------------------------------------------------------
**Error 1**
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[50], line 2
      1 # export spike sorting report to a folder
----> 2 si.export_report(analyzer_clean, base_folder / 'report_Test_KS1', format='png')
File ~\Documents\Domi\Python\spikeinterface\src\spikeinterface\exporters\report.py:144, in export_report(sorting_analyzer, output_folder, remove_if_exists, format, show_figures, peak_sign, force_computation, **job_kwargs)
    [139](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:139) for unit_id in unit_ids:
    [140](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:140)     fig = plt.figure(
    [141](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:141)         constrained_layout=False,
    [142](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:142)         figsize=(15, 7),
    [143](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:143)     )
--> [144](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:144)     sw.plot_unit_summary(sorting_analyzer, unit_id, figure=fig)
    [145](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:145)     fig.suptitle(f"unit {unit_id}")
    [146](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/exporters/report.py:146)     fig.savefig(units_folder / f"{unit_id}.{format}")
File ~\Documents\Domi\Python\spikeinterface\src\spikeinterface\widgets\unit_summary.py:60, in UnitSummaryWidget.__init__(self, sorting_analyzer, unit_id, unit_colors, sparsity, radius_um, backend, **backend_kwargs)
     [51](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:51)     unit_colors = get_unit_colors(sorting_analyzer.sorting)
     [53](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:53) plot_data = dict(
     [54](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:54)     sorting_analyzer=sorting_analyzer,
     [55](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:55)     unit_id=unit_id,
     [56](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:56)     unit_colors=unit_colors,
     [57](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:57)     sparsity=sparsity,
     [58](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:58) )
---> [60](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:60) BaseWidget.__init__(self, plot_data, backend=backend, **backend_kwargs)
File ~\Documents\Domi\Python\spikeinterface\src\spikeinterface\widgets\base.py:87, in BaseWidget.__init__(self, data_plot, backend, immediate_plot, **backend_kwargs)
     [84](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:84) self.backend_kwargs = backend_kwargs_
     [86](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:86) if immediate_plot:
---> [87](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:87)     self.do_plot()
File ~\Documents\Domi\Python\spikeinterface\src\spikeinterface\widgets\base.py:108, in BaseWidget.do_plot(self)
    [106](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:106) def do_plot(self):
    [107](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:107)     func = getattr(self, f"plot_{self.backend}")
--> [108](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/base.py:108)     func(self.data_plot, **self.backend_kwargs)
File ~\Documents\Domi\Python\spikeinterface\src\spikeinterface\widgets\unit_summary.py:107, in UnitSummaryWidget.plot_matplotlib(self, data_plot, **backend_kwargs)
    [105](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:105) unit_location = unit_locations[unit_id]
    [106](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/cashlab/Documents/Domi/Python/spikeinterface/examples/how_to/~/Documents/Domi/Python/spikeinterface/src/spikeinterface/widgets/unit_summary.py:106) x, y = unit_location[0], unit_location[1]
[... cut ...]
    [378](file:///C:/Users/cashlab/miniconda3/envs/si_env/lib/site-packages/simplejson/encoder.py:378) try:
--> [379](file:///C:/Users/cashlab/miniconda3/envs/si_env/lib/site-packages/simplejson/encoder.py:379)     return _iterencode(o, 0)
    [380](file:///C:/Users/cashlab/miniconda3/envs/si_env/lib/site-packages/simplejson/encoder.py:380) finally:
    [381](file:///C:/Users/cashlab/miniconda3/envs/si_env/lib/site-packages/simplejson/encoder.py:381)     key_memo.clear()

**ValueError: Out of range float values are not JSON compliant**

alejoe91, 2024-05-14:
Hi Domi,

I think that the problems i that you're getting `NaN` estimates of unit locations, since the error comes from 
```x, y = unit_location[0], unit_location[1]
--> spikeinterface/src/spikeinterface/widgets/unit_summary.py:107) ax1.set_xlim(x - 80, x + 80)
```
We should definitely protect against something like this interfering with plotting. I'll make a PR soon :)

zm711, 2024-12-03:
Hey @alejoe91 did we fix this?

@meszena could you check with the code on main to see if we fixed this :)

closed 2025-12-03 by alejoe91