SpikeInterface/spikeinterface #2847: Verbose clashing in `recording.save_to_folder` and `job_kwargs`

h-mayorquin, 2024-05-14:
As in the title, 

https://github.com/SpikeInterface/spikeinterface/blob/3864afde8aad572041b4218ebb1d50b2e7a58d6b/src/spikeinterface/core/base.py#L849-L850

There is a verbose in `save_to_folder` but it also clashes with the `save_kwargs` which should contain `job_kwargs`. 

Suggestion: drop the verbose in `save_to_folder` or rename it to `print_used_folder` if someone wants that functionality.

I swear I read about a similar issue but could not find it.

zm711, 2024-05-14:
Also related #2728, no?

samuelgarcia, 2024-05-14:
I want to remove verbose from job_kwargs!

h-mayorquin, 2024-05-14:
> I want to remove verbose from job_kwargs!

What does that mean? Removing the argument from the `ChunkRecordingExecutor`, removing it from global?

samuelgarcia, 2024-05-15:
from the global because verbose is used randomly at many other place with is not necessarly form parallel stuff.

h-mayorquin, 2024-05-15:
> from the global because verbose is used randomly at many other place with is not necessarly form parallel stuff.

Got you

h-mayorquin, 2024-05-16:
But thinking about it, how would that solve the problem in this issue? How would you propagate the verbose all the way down to the initialization of the `ChunkRecordingExecutor` if not in the `job_kwargs`? And as long as that is the way of passing the verbose in the job_kwargs it will clash with the verbose of other signatures.

That seems independent of whether it is set globally or not. Am I missing something? Just thinking aloud here.

h-mayorquin, 2024-05-16:
I think that to avoid clashing we should just create another variable that is called `parallel_verbose` ...

PR #2901 (2024-05-24): Fix remaining verbose clashing
Should close #2847. Following on https://github.com/SpikeInterface/spikeinterface/pull/2898/files. Also removed some comments from https://github.com/SpikeInterface/spikeinterface/pull/2728.

I still think this approach of catching verbose at every point of the call stack is very labor intensive. I mentioned to Sam that we could have more specific verbosity terms:

For the ChunkRecordingExectuor: `display_chunk_info` or `verbose_chunk_executor` 
For the save_to_folder `display_saved_folder`

But he pointed out -correctly I think- that usually verbose is managed i [... cut]
closed 2024-05-24 by alejoe91