SpikeInterface/spikeinterface #2849: Link with random spikes and waveforms extension

Antho2422, 2024-05-15:
Hi SpikeInterface, 

I have a question about the random spikes extension. I have a SortingAnalyzer with 5 units but my random spike extension has shape (n_random_spikes,).  I expected random spikes to have shape (n_random_spikes, 5) (i.e each timing of the spikes for each unit).

So I don't really know how to retrieve the timings of each waveforms for each unit in the waveforms extension... 

Can somebody explain me how I should use random spikes to retrieve waveforms times depending on the unit ? 

Thank you, 
Anthony

samuelgarcia, 2024-05-15:
Salut Antony,
random_spikes give you the indices of selected sikes the full spike vector.

```python
# this is the full vector of spikes
spikes = sorting.to_spike_vector()
# this is indicies
random_spikes_indices = analyzer.get_extension().get_data()
# this is selected spikes
selected_spikes = spikes[random_spikes_indices]
```

Antho2422, 2024-05-15:
Salut Samuel, 

Thanks, I understand now thank you !

closed 2024-05-15 by Antho2422