SpikeInterface/spikeinterface #2855: Additional checks on GPU when running containerised sorting

JoeZiminski, 2024-05-15:
I often have a hard time debugging GPU issues when running containerised sorting. There are already a lot of useful checks and it is not straightforward e.g. #1398 as can't always rely on commands like `nvidia-smi` being accessible. I've divided this into 'easy checks' and 'harder checks'.

### Easier Checks

If would be good to check that docker or singularity is installed at all on the system at all (e.g. `docker --version` returns nonzero code). At the moment if not installed it gives a unclear error. 

If you are running singularity `spython` is a dependency and if docker then `docker` (downloaded from PyPi). However these can't be installed easily by default as they don't all work cross platform. You could do something like the below in the `pyproject.toml`

```
"docker; platform_system=='Windows'",
"docker; platform_system=='Darwin'",
"spython; platform_system=='Linux'",  # I think missing from SI?
"cuda-python; platform_system != 'Darwin'",
```

Alternatively, it would be nice to raise an error if trying to run with docker and `docker` is not installed or running with singularity and `spython` is not installed. 

### Harder Checks

Today we had an issue that was quite hard to debug where  `nvidia-docker` was not installed but was required. It would be great to check if running on docker `nvidia-docker` is installed (some details [here]). However, it sounds like this is a bit of a nightmare, AFAIK it's not cross-platform and [nvidia-docker](https://github.com/NVIDIA/nvidia-docker?tab=readme-ov-file) is now superseded by [nvidia-container-toolkit](https://github.com/NVIDIA/nvidia-container-toolkit). Nonetheless it might be worth checking that if trying to use docker, and on Linux, that either `nvidia-docker` or `nvidia-docker2` or `nvidia-docker-toolkit` is installed (some more detailed explanation between their differences [here](https://github.com/NVIDIA/nvidia-docker/issues/1268).

https://docs.docker.com/desktop/gpu/

h-mayorquin, 2024-05-15:
I think there are some hanging fruit that you pointed out. That's great! We should make a TODO list : )

h-mayorquin, 2024-05-16:
Great! Thanks for the actionable plan.

alejoe91, 2024-05-23:
@JoeZiminski this could be another "quick" project for the hackahton! Would you like to add it to the list of projects?

JoeZiminski, 2024-06-12:
Thanks both, sorry @alejoe91 missed this prior to the hackathon!

I will open a PR for (1) and (2) now. I think for (3), because I am not 100% sure exactly how it will work and am worried about raising an error, how about instead raising a warning? e.g. if they these dependencies are not found on a linux machine when using docker, something like `nvidia-docker`, `nvidia-docker2` or `nvidia-docker-toolkit` not found, this may cause an error when running GPU-dependent dockers. Install XXX (not exactly sure which one, I think the toolkit is most recent) to fix this error.'

PR #3024 (2024-06-28): Add more container dependency checks in `run_sorter`
This PR closes #2855 by adding in additional checks on container and GPU dependencies when running `run_sorter()`. 

It adds checks that `docker` or `singularity` are installed (if running sorter with `docker` or `singularity` respectively) and also that the corresponding python packages `docker` and `spython` are installed. It also show a warning if, on Linux, some nvidia-docker tools are not available, which  are usually necessary. See #2855 for a disucssion, but these nvidia-docker dependencies are a little complex so instead of raising a [... cut]
closed 2024-06-28 by samuelgarcia