SpikeInterface/spikeinterface #2863: Pin version of matplotlib

h-mayorquin, 2024-05-16:
Also from #2861. 

We are getting the following error:

```python

E       AttributeError: module 'matplotlib.cm' has no attribute 'get_cmap'

```

We need to figure out when this attribute was introduced or deprecated. The version used is 3.9 there https://github.com/SpikeInterface/spikeinterface/actions/runs/9112219162/job/25051018131?pr=2861 so most likely a deprecation.

h-mayorquin, 2024-05-16:
Ok, is the latest version that was just released:

https://matplotlib.org/stable/api/prev_api_changes/api_changes_3.9.0.html#top-level-cmap-registration-and-access-functions-in-mpl-cm

I am not familiar with plotting. I think it is up to @alejoe91 and @samuelgarcia if they want to add a upper bound or change the code to work with matplotlib latest. If they are too busy and this does not move on on the following days I will put a ceiling on the version so avoid failing tests and then we can revert once someone has time to update the code.

h-mayorquin, 2024-05-16:
Yes, it is more than this is a busy time for us. 

I have never seen testing of matplotlib that I feel is satisfactory. Automatizing plot output testing seems like a difficult thing to do (see the testing suit of ipywidgets). Have you seen or done that in a way that feels good?

JoeZiminski, 2024-05-16:
Yes good point about it being a busy time, how about we fix this one, but if it is leading to a lot of errors we can pin until things are less busy?

Yes I agree testing plots is always unsatisfying, you can never be sure the plot is rendering visually as you'd expect. I think the best you can do is to check that the expected axes, axes limits, titles and data are shown on the plot across all possible input arguments. For example I [added some tests recently](https://github.com/SpikeInterface/spikeinterface/blob/394203210aeb20015eae8a293c9e6f9479052017/src/spikeinterface/widgets/tests/test_motion_widgets.py) for testing `plot_motion` that tests some of these things. At the end of the day you cannot be sure it is rendering as expected, but if the API is showing it is then that is a problem on the matplotlib side I guess. It's kind of a similar problem to testing GUIs, the [textual test suite](https://github.com/Textualize/textual/tree/main/tests) has some nice examples of this.

h-mayorquin, 2024-05-21:
I will keep this open until Sam updates the issue just as a todo.

h-mayorquin, 2024-05-30:
This  was solved by https://github.com/SpikeInterface/spikeinterface/pull/2891

closed 2024-05-30 by h-mayorquin