SpikeInterface/spikeinterface #2865: Question MergeUnitSorting unit_ids

Antho2422, 2024-05-16:
Hi, 

I have a question about the merge performed using `MergeUnitSorting`. 

If `sorting.get_unit_ids() = [0, 1, 2, 3]` and I merge unit 0 and 1 together the output will be `new_sorting.get_unit_ids() = [2, 3, 4]` right ? 

Is it always the case and if not, is there a way to make sure of which merge resulted in which unit ? 

I want to make sure that I'm looking to the right units when a merge is performed.

Thanks,
Anthony

Antho2422, 2024-05-16:
I also noticed some weird things when plotting the templates of the units I merge and the merged result. It seems like the waveforms is shifted on the right, I don't know what could be the reason because the number of samples of the templates is the same. 

Here is an example of what I mean (the red unit is, if the above is right, the merged unit): 
[image]

JoeZiminski, 2024-05-17:
Hi Anthony, it seems the default behaviour of `merge` is to use the next number than has never been used for an id before (see [here](https://github.com/SpikeInterface/spikeinterface/blob/024c24afaec2d0a0476017c77a75f7c8a1dcb4b3/src/spikeinterface/curation/curationsorting.py#L137)) and [here](https://github.com/SpikeInterface/spikeinterface/blob/024c24afaec2d0a0476017c77a75f7c8a1dcb4b3/src/spikeinterface/curation/curationsorting.py#L57). If for example you merge 2,3 the new unit will be 5 (it will not go back to 1 or 2).  However, you can overwrite this behaviour by specifying `new_unit_id` (see end of the example below).

That is interesting about the merged plots, could you please post the code you are using (or a minimal reproducable example)? The test code below seems to look okay:

```
import spikeinterface as si
import matplotlib.pyplot as plt
import spikeinterface.curation as si_curation
import numpy as np

# Generate curated sorting
recording, sorting = si.generate_ground_truth_recording()
curation_sorting = cs = si_curation.CurationSorting(parent_sorting=sorting)

# first extract the templates for units 0 and 1
waveforms = si.extract_waveforms(recording, curation_sorting.sorting, mode="memory")

print(waveforms.sorting.get_unit_ids())
# [0 1 2 3 4 5 6 7 8 9]

unit_0 = waveforms.get_template(unit_id=0)
unit_1 = waveforms.get_template(unit_id=1)

# Then merge zero and one, and extract the new merged template (unit 10)
curation_sorting = si_curation.CurationSorting(parent_sorting=waveforms.sorting)
curation_sorting.merge(units_to_merge=[0, 1])
print(curation_sorting.sorting.get_unit_ids())
# [ 2  3  4  5  6  7  8  9 10]

curated_waveforms = si.extract_waveforms(recording, curation_sorting.sorting, mode="memory")
merged_0_and_1 = curated_waveforms.get_template(unit_id=10)

plt.plot(np.average(unit_0, axis=1))
plt.plot(np.average(unit_1, axis=1))
plt.plot(np.average(merged_0_and_1, axis=1))
plt.legend(["0", "1", "merged"])

plt.show()

curation_sorting.merge(units_to_merge=[2,3])
print(curation_sorting.sorting.get_unit_ids())
# [ 4  5  6  7  8  9 10 11]

curation_sorting.merge(units_to_merge=[4,5], new_unit_id=0)
print(curation_sorting.sorting.get_unit_ids())
# [ 6  7  8  9 10 11  0]
```

Antho2422, 2024-05-23:
@JoeZiminski 

Hey, thanks a lot for your answer. For the unit ids it is much more clear now thanks a lot. 

About the plotting issue, I need to double check, this may come from my side. I will rework on that when I will have some time but seeing your example it seems to work fine so it may come from my code. 

Thank you a lot, 

Anthony

zm711, 2024-06-19:
I'll close this one for now. But if you still see the problem just reopen.

closed 2024-06-19 by zm711