SpikeInterface/spikeinterface #2869: detect_bad_channels is really slow

WeissShahaf, 2024-05-17:
the longest part of preprocessing is the detect_bad_channels part. even if i slice the recording to just 30 seconds (384 channels)
can it be optimized? perhaps using parallelization? 

thanks

h-mayorquin, 2024-05-17:
I took a brief look a the function and that makes sense.  

Which method are you using? Are all of them equally slow?

https://github.com/h-mayorquin/spikeinterface/blob/bc0cbd9f74f30cfed9376f984091cd57f3634169/src/spikeinterface/preprocessing/detect_bad_channels.py#L124-L125

Also, if you could investigate what is the bottleneck with a profiler we could come up with a lan.

alejoe91, 2024-05-20:
Thanks! I'll give it a stab to parallelize it soon

closed 2026-09-08 by alejoe91