SpikeInterface/spikeinterface #2871: Should `get_traces` with `return_scaled=True` when the base dtype is float.

h-mayorquin, 2024-05-17:
From the discussion inhttps://github.com/SpikeInterface/spikeinterface/pull/2864

We have the following lines:

https://github.com/h-mayorquin/spikeinterface/blob/bc0cbd9f74f30cfed9376f984091cd57f3634169/src/spikeinterface/core/baserecording.py#L369-L373

But if the original dtype is float64 this will downcast. Maybe we should change the code a little bit.

JoeZiminski, 2024-05-17:
Agree it makes sense if the dtype of the recording is already `float64` to have `get_traces` return `float64` when `get_traces` is used. So overall if I understand SI behaviour correctly the (new) system would be:

Original          Unscaled.         Scaled
`int`                  `int`              `float32`
`float32`       `float32 `        `float32`
`float64 `      `float64 `        `float64 `

h-mayorquin, 2024-05-17:
Yes, int should be cast to float.

 I think that float should not be cast at all. Maybe float16 if that appears for some reason? Does it appear?

JoeZiminski, 2024-05-20:
Good point, yes `float16` seems to be [supported](https://github.com/search?q=repo%3ASpikeInterface%2Fspikeinterface%20float16&type=code) and would agree it makes sense that the `dtype` for any float should be preserved between `return_scaled` and `return_unsclaed`.