SpikeInterface/spikeinterface #2883: Commented out code for debugging fitting collissions

JoeZiminski, 2024-05-19:
Small thing but there is is some commented out code for debugging the spike collisions [here](https://github.com/SpikeInterface/spikeinterface/blob/024c24afaec2d0a0476017c77a75f7c8a1dcb4b3/src/spikeinterface/postprocessing/amplitude_scalings.py#L535). It would be nicer if this code was uncommented and made non-public with a leading underscore. Then their purpose for debugging can be explained in the docstring.

alejoe91, 2024-05-19:
Hey @JoeZiminski 

I can extend the docs and uncomment the functions in fixing #2882

JoeZiminski, 2024-05-19:
Brilliant thanks @alejoe91 I was looking at #2882 that made me think about this, let me know if I can help with anything!

alejoe91, 2024-05-19:
Well if you have time you can have a stab! My plan was just to protect the `LinearRegression` function and set scalings to NaN for these spikes.

Regarding what this function does:
The idea is fitting templates to each spike and return the amplitude scaling. This is better than spike_amplitudes, which simply take the voltage value at the spike time, since the latter fails when there is a collision. In addition, also without a collision, fitting the template to the traces snippet is a more robust estimate of amplitude.

JoeZiminski, 2024-05-20:
Sure I'll have a go!

JoeZiminski, 2024-05-21:
Adressed in #2893