SpikeInterface/spikeinterface #2886: Type Hints for boolean variables

zm711, 2024-05-20:
I know type hints are still a little contentious, but I'm wondering if we can all agree for type hints on all boolean variables moving forward as part of development rather than as an afterthought.

For example the argument `output` is used in the code base as a string `"dict"` or `'id'`, but could just as easily be interpreted as a boolean like `output=True` meaning I want to return a value. So if we agree to mark `:bool` moving forward I think it will help us read these arguments and if new developers come along they will also have a clearer idea of some of these arguments that could easily be misinterpreted. 

Anyone really opposed to requesting boolean type hints?

This was inspired by @h-mayorquin's PR #2868, where I realized that I wasn't sure if `return_x` was always a boolean in the code base or if sometimes it would also be a string specifying how to return something (e.g. vector vs dict).

h-mayorquin, 2024-05-20:
Yes, I think specially if we have want to have very laconic / concrete varialbe naming then typing becomes really useful for code readers. 

There is also output vs outputs somewhere there.

JoeZiminski, 2024-05-20:
Big +1 on this, static type checking is fantastic protection against bugs, I feel dangerous performing refactoring without it now. For me anything that gets closer to a fully typed codebase (and a type-checker in the linting) is a positive!

samuelgarcia, 2024-05-21:
Sometimes, the output can have 3 possibility no ?

zm711, 2024-05-21:
It fine for me for it to have 3 possibility, but then I prefer

```
output: bool | 'id' | 'dict'
```

or with typing from std library

```
output: bool | Literal['id'|'dict']
```

It makes debugging a bit easier to know what the goal of the argument is, so I'm not saying we have to reduce the input necessarily just that we document what the arguments ought to be with a focus on making sure we know when arguments are just boolean flags vs actual values.

h-mayorquin, 2024-05-21:
> Sometimes, the output can have 3 possibility no ?

True. I don't like this when this happens : P

zm711, 2024-06-21:
I'll close this one since the discussion is pretty much as expected. :)

closed 2024-06-21 by zm711

h-mayorquin, 2024-06-21:
Isn't there something actionable we can do out of here?

JoeZiminski, 2024-06-21:
I'm a strong advocate for adding `mypy` as a static type checker to the pre-commit hooks, even without much type hinting in the codebase. Without type hints, or with minimal type hints, it will just provide very useful warnings when there are egregious issues in the code (e.g. function called with wrong argument names). If type hints are slowly added, it will slowly catch more errors. For typing basic things like booleans, `mypy` shouldn't cause any frustration day-to-day and should only find real issues in the code base that need fixing. These issues can either be caught by the linter or caught by a user who creates an issue, so either way they will have to be fixed, and its less pain to do it during development. I think at least opening a PR to see the kinds of things `mypy` catches would be useful (I'd be happy to do next week).

zm711, 2024-06-21:
> Isn't there something actionable we can do out of here?

I didn't open as an "action" issue, but more to make sure Sam knew that we all love typing. You have to slip it in small doses to encourage good behavior.

> I think at least opening a PR to see the kinds of things mypy catches would be useful (I'd be happy to do next week).

Just do it after Sam has gone to bed. Then we have hours to see.

JoeZiminski, 2024-06-21:
😆 There is the road to Damascus conversion moment when the linter catches something and you think 'oh gosh that would have been a really nasty bug!'. But for that to happen the linter needs to be snuck into the pre-commit overnight 😈

samuelgarcia, 2024-06-25:
> Just do it after Sam has gone to bed. Then we have hours to see.

C'est une comspiration! A l'aide!