SpikeInterface/spikeinterface #2887: `SortingAnalyzer` `.compute("spike_amplitudes") error

JoeZiminski, 2024-05-20:
I'm running into trouble with the below code:

```
import spikeinterface as si

# Generate curated sorting
recording, sorting = si.generate_ground_truth_recording()  # num_channels=64

sorting_analyzer = si.create_sorting_analyzer(
    sorting, recording
)

sorting_analyzer.compute("random_spikes")
sorting_analyzer.compute("waveforms")
sorting_analyzer.compute("templates")

sorting_analyzer.compute(
    "spike_amplitudes",
    peak_sign="neg"  # "pos" does not work either"
)

# sorting_analyzer.compute("amplitude_scalings")
```

which gives an error:

```
Traceback (most recent call last):
  File "/Users/joeziminski/PycharmProjects/si-dev/main.py", line 14, in <module>
    sorting_analyzer.compute(
  File "/Users/joeziminski/git_repos/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 869, in compute
    return self.compute_one_extension(extension_name=input, save=save, **kwargs)
  File "/Users/joeziminski/git_repos/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 924, in compute_one_extension
    for child in _get_children_dependencies(extension_name):
  File "/Users/joeziminski/git_repos/spikeinterface/src/spikeinterface/core/sortinganalyzer.py", line 1217, in _get_children_dependencies
    children = _extension_children[extension_name]
KeyError: 'spike_amplitudes'
```

I'm following from [here](https://spikeinterface.readthedocs.io/en/latest/tutorials/waveform_extractor_to_sorting_analyzer.html), apologies if I have missed something!

zm711, 2024-05-20:
@JoeZiminski, could you try restarting your kernel. I'm not 100% certain this is the issue, but the way the extensions are generated can lead to an issue if you change code partway through analysis. Just curious is restarting with a clean kernel fixes this issue.

I just tested your code with a fresh ipython session and it worked fine.

JoeZiminski, 2024-05-20:
Thanks @zm711 I am also getting this in a fresh kernel / running as a script.  I am on macOS m2, I'll try and replicate on windows in a fresh environment, no idea if OS should matter but I can't think of anything else at this stage 🤔

zm711, 2024-05-20:
I figured it out @JoeZiminski :).

When I tested I did `import spikeinterface.full as si` instead of just importing core. You are not import the postprocessing package so it's not working. 

I thought we had a better error for this/automatic loading, which seems to not be working anymore.

could you try re-running with a full import? And if that works then I think we need to improve the error messaging.

zm711, 2024-05-20:
@alejoe91 and @samuelgarcia 

The issue is this:
https://github.com/SpikeInterface/spikeinterface/blob/024c24afaec2d0a0476017c77a75f7c8a1dcb4b3/src/spikeinterface/core/sortinganalyzer.py#L924-L928

We try to delete before we "get" which means that non-core extensions don't work and we get an unhelpful error message. I'll try to open a PR switching the order and seeing if breaks anything unexpected.

JoeZiminski, 2024-05-20:
Thanks @zm711! that did the trick. I had a look around the `sorting_analyzer.py` but couldn't figure it out thanks for tracking down the root cause. I say this with absolutely no understanding of the requirements of the class and nature of the problem, but is it possible to avoid the global variables used in `sorting_analyzer.py`? Not sure if it's the case here but they are often the cause of many bugs and makes it difficult to follow the logic of the code.

zm711, 2024-05-20:
That's why I tagged Sam and Alessio. That's above my paygrade :) 

I did a draft PR to try to fix it, but I'm waiting to see if there are unintended consequences of this. Basically an auto-load was added just for the extensions that gets triggered by get_extension but since we were trying to clean-up before we had the extension registered we were getting a key error. If you already import the packages then they are registered at import so this isn't a problem (and why we hadn't discovered it yet is my guess).

JoeZiminski, 2024-05-20:
That's brilliant, thanks for solving this so fast. Is it worth to include this MRE (or something similar) as a quick unit test in `test_sortinganalyzer`?

zm711, 2024-05-20:
Yeah it probably is worth adding, I'll have to look at the tests though. Because once we've imported it might be too late. I'll have to see what the file looks like. I know adding more to the test suite is a big push (and will be part of the hackathon) so hopefully you can help out with !

PR #2888 (2024-05-21): Put get_extension (to trigger auto-import) before get_children_dependencies for deleting Analyzer extensions to prevent KeyError
Fixes #2887 

## Issue

See #2887. But the issue is delete on recompute won't work for non-core extensions since the get_child_dependencies won't have the other extensions registered. We either need to flip the order (which I want to test with this PR) or fix the error messaging for this case.

Logic is that getting the class_extension should register it, which would allow us to delete on recompute.
closed 2024-05-21 by samuelgarcia