SpikeInterface/spikeinterface #2890: Missing spikes

simonarvin, 2024-05-21:
Hi!

I have a 1 minute single-channel recording from a human patient which I'm processing using SpikeInterface.

The spike sorting runs without errors, but it detects only 1 cluster. The picture below shows the output in the SpikeInterface GUI; I presume the dots indicate spikes assigned to the cluster, and that the other spikes could not be assigned to a cluster.

[image]

Any ideas why so many spikes remain unassigned? Are they not detected as spikes in the first place? We worry that we miss out on other potential clusters. I have tried SpykingCircus2, Tridesclous and MountainSort5 with essentially the same results.

The cluster we do find looks good:
[image]

Below is the function I use to do spike sorting with SpykingCircus2:
```python
self.sorting = ss.run_sorter(
                "spykingcircus2",
                self.recordings[recording_key],
                output_folder = f"{src}/spikesort-spykingcircus2-c{self.channel}-{self.timestamp}",
                verbose = True,
               apply_preprocessing = False,
                detection={"peak_sign" : "pos", "radius_um" : 100},
                clustering={
                    "smart_search": True,
                    "cc_merge": 1,
                },
            )
```

The parameters (`smart_search`, `cc_merge`) are recommended in the original SpykingCircus documentation for single channel recordings.

The data is bandpass filtered (`{"freq_min": 300, "freq_max": 8000} `)

Best,
Simon

zm711, 2024-05-21:
Hey Simon,

I think the best people to comment on TDC and SC1/2 are @samuelgarcia and @yger, respectively. 

@magland would be able to comment on whether he thinks ms5 would work on single channel data.

magland, 2024-05-21:
Yes, ms5 should be able to handle single channel data.

samuelgarcia, 2024-05-22:
Hi Simon,
missing spikes looks like a bug!
Note that TDC2 is still work-in-progress.

Maybe you could try the run_sorter('simple', ...) which is a bit simple but for simple case with unique channels should relevant.

simonarvin, 2024-05-22:
Thank you, all!

@samuelgarcia: Does the "simple" sorter have any documentation material? I couldn't find it anywhere on SpikeInterface's ReadTheDocs page.

I tried running:
```python
ss.get_default_sorter_params(sorter_name_or_class='simple')
```
which lists the sorter's parameters, but the description getter returns an empty dictionary:
```python
ss.get_sorter_params_description(sorter_name_or_class='simple')
```

When I run the "simple" sorter with default parameters, the sorting performs very poorly (it misses a lot of spikes while detecting noise as spikes). Interestingly, the recording has a very good SNR, so the detected false spikes in the noise are actually of much lower amplitude than the actual spikes that are not detected.

Best,
Simon

zm711, 2024-05-22:
It is not documented yet. But we should!

You could post here if you have a specific question about a param until we update the documentation!

simonarvin, 2024-05-23:
Thank you for the incredible support!
Got it to work with a bit more fiddling with MountainSort5.

closed 2024-05-23 by simonarvin

alejoe91, 2024-05-23:
@simonarvin is your data inverted? I mean, so you expect spikes to have positive peaks? In that case you should set the `peak_sign` arguments to `"pos"` in the simple sorter params, same for SC2 and TDC2 ;)

Glad that MS5 works well!

rat-h, 2024-05-23:
@samuelgarcia is the simple sorter in the latest pip version?
```python
import numpy as np
from probeinterface import read_probeinterface
import spikeinterface.full as si

print(si.__version__)
recording = si.BinaryRecordingExtractor("continuous.dat",30000.0,'int16', num_channels=128)
recording = recording.remove_channels([])
prob = read_probeinterface("probes/A4x32-Poly2-5mm-23s-200-177-after-mapping.json").probes[0]
recording.set_probe(prob,in_place=True)

preproc = si.filter(recording, btype="bandpass",band=[300,6000])
preproc = preproc.save( folder="si2simple/preprocessed", chunk_duration='60s',overwrite=True)

sorting = si.run_sorter(sorter_name='simple',recording=preproc, output_folder="si2simple/sorting-workingdir")
sorting== sorting.save(folder="si2simple/s/sorting-saved",chunk_duration='60s')
```

```
python si2simple.py 
0.100.6
write_binary_recording with n_jobs = 1 and chunk_size = 1800000
write_binary_recording: 100%|##############################################################################################################################################################################| 17/17 [02:43<00:00,  9.64s/it]
Traceback (most recent call last):
  File "/home/rth/spikeinterface/si2simple.py", line 14, in <module>
    sorting = si.run_sorter(sorter_name='simple',recording=preproc, output_folder="si2simple/sorting-workingdir")
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 175, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 219, in run_sorter_local
    SorterClass = sorter_dict[sorter_name]
KeyError: 'simple'
```

rat-h, 2024-05-23:
I manually add simple from github (not a good idea, I know) just to try, and it returns an error
```
Traceback (most recent call last):
  File "/home/rth/spikeinterface/si2simple.py", line 15, in <module>
    sorting = si.run_sorter(sorter_name='simple',recording=preproc, output_folder="si2simple/sorting-workingdir")
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 175, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 225, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 293, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
concurrent.futures.process._RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 246, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 205, in _process_chunk
    return [fn(*args) for args in chunk]
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 205, in <listcomp>
    return [fn(*args) for args in chunk]
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 439, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/node_pipeline.py", line 522, in _compute_peak_pipeline_chunk
    node_output = node.compute(traces_chunk, *node_input_args)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sortingcomponents/waveforms/temporal_pca.py", line 232, in compute
    projected_temporal_waveforms = self.pca_model.transform(temporal_waveforms)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/utils/_set_output.py", line 142, in wrapped
    data_to_wrap = f(self, X, *args, **kwargs)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/decomposition/_truncated_svd.py", line 287, in transform
    X = self._validate_data(X, accept_sparse=["csr", "csc"], reset=False)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/base.py", line 535, in _validate_data
    X = check_array(X, input_name="X", **check_params)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/utils/validation.py", line 929, in check_array
    raise ValueError(
ValueError: Found array with 0 sample(s) (shape=(0, 75)) while a minimum of 1 is required by TruncatedSVD.
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/internal/simplesorter.py", line 161, in _run_from_folder
    output = run_node_pipeline(
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/node_pipeline.py", line 471, in run_node_pipeline
    processor.run()
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 401, in run
    for res in results:
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 575, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 621, in result_iterator
    yield _result_or_cancel(fs.pop())
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 319, in _result_or_cancel
    return fut.result(timeout)
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 451, in result
    return self.__get_result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
ValueError: Found array with 0 sample(s) (shape=(0, 75)) while a minimum of 1 is required by TruncatedSVD.

Spike sorting failed. You can inspect the runtime trace in /home/rth/spikeinterface/si2simple/sorting-workingdir/spikeinterface_log.json.
```

h-mayorquin, 2024-05-23:
> Thank you for the incredible support! Got it to work with a bit more fiddling with MountainSort5.

It would be useful for future users if you mention fiddling fix it. Could you add the final set of parameters for example?