SpikeInterface/spikeinterface #2897: Can read_tdt() handle .sev files?

mberns-ru, 2024-05-22:
Hi! Currently, I am trying to load data into a recording object like this:
`full_raw_rec = si.read_tdt(folder_path=tdt_folder, stream_name = "b'Wav1'")`

where _tdt_folder_ has the following structure:
[image]

We use a 64-channel probe, and each channel is exported into a .sev file. However, when I pass in _tdt_folder_ to read_tdt(), I get the following error:
[image]

My understanding - based on Issue #2425 - is that read_tdt() is looking for a .tev file. When I pass in that file directly, it does work; however, our .tev file doesn't include the 64 channels that we want, which are stored in 64 separate .sev files. It seems that Neo's [TdtRawIO function](https://neo.readthedocs.io/en/0.13.1/rawiolist.html#neo.rawio.TdtRawIO) does support .sev files.

I'm unsure how to go about loading these .sev files into a single recording object. I was wondering if anyone could provide some guidance? I apologize if this is a silly question; I'm new to this!

zm711, 2024-05-23:
Maybe related: 
https://github.com/NeuralEnsemble/python-neo/issues/1286

Is this a newer style set of `.sev` files. If so we may want to open an issue on Neo and if you can supply some sample files or a folder we can work on fixing that over there. I don't know tdt at all, but with sample files I can at least try to see what is going on.

victornovakov, 2024-05-23:
Hi I recently had this problem as well. Read tdt can read the entire folder which is created including all the sev files as well as the tev and other files. Use the filepath to the folder

h-mayorquin, 2024-05-23:
@victornovakov does that mean that https://github.com/SpikeInterface/spikeinterface/issues/2825 worked for you? Can we close it?

victornovakov, 2024-05-23:
@h-mayorquin yes it can be closed. Sorry I forgot to follow up on the issue.

h-mayorquin, 2024-05-23:
No worries, it is good that you asked so when somebody googles it might land them there.

mberns-ru, 2024-05-23:
@zm711 Sorry for the delay! I commented on that Neo post with a sample dataset.

zm711, 2024-05-24:
Hi @mberns-ru!

So your error is actually due to the fact that your dataset is a single TDT block rather than a multi-block dataset. So what you need to do is:

```python
tdt_folder = r'Path/to/folder'
full_raw_rec = si.read_tdt(folder_path=tdt_folder / 'A_Only_Test_Ephys_Passive_HL-240514-091335_J_M41_Tail-240523-094748.Tbk' , stream_name = "b'Wav1'")
```
Notice that I took your folder path and added on one file inside the folder path.
For single-block you have to input any file to get the directory. For multi-block you have to put in the root directory. 

I think this is actually pretty confusing. Is there a reason for this @samuelgarcia (you wrote the IO originally)?

mberns-ru, 2024-05-24:
@alejoe91 Thank you so much!