SpikeInterface/spikeinterface #2913: detect_and_localize is not using the GPU

JuanPimientoCaicedo, 2024-05-28:
Hi, guys. 

I hope everyone is doing well. I wonder if this is normal behavior of the correct_motion function. I believe it should use the GPU for the detect_and_localize step. But this is what I am getting. When running the code, the CPU get's totally.

[image]

The drift correction works well, the only problem is the processing time.

Here you can see the drift correction. 

[image]

'''
print(torch.cuda.is_available()) 
print(si.__ version __)

  True
  0.101.0

# Calculate drift traces

motion_folder =  Path(new_folder) /'motion_correction'  

job_kwargs = {'n_jobs':-1, 
              'chunk_duration':'1s', 
              'progress_bar': True}

rec_corrected = si.correct_motion(recording= drift_recording_cmr, preset="nonrigid_accurate", 
                                  folder=motion_folder,  **job_kwargs)
'''

Am I missing an argument to recruit the GPU? By the way, when I spikesort, the GPU does get recruited.

Thank you for your help.

samuelgarcia, 2024-06-03:
Hi,
detect and localize is only CPU coded in spikeinterface. But should be quite fast normally with our numba kernel.
What is the duration of the recording ?
On a  20 core cpu machine this process is faster than the duration of the recording on our side but this depend on many factor like the data Io access.

JuanPimientoCaicedo, 2024-06-03:
Hello,
I understand. The recording is indeed a little long, about 4 hours. It makes sense that it takes long. Additionally, my CPU is currently a 11th Gen Intel Core i7-11700K-8 core CPU. I guess I will leave it running overnight.

You can close this issue if you want,

Thank you!

zm711, 2024-06-04:
I think what Sam was emphasizing is that if you are trying to do this file over network or hard drive rather than locally you could run into big io bottlenecks. If you're not trying this locally then you should try it locally to help assess io bottlenecks. 

We can close this for now, but if something related to this comes up then feel free to reopen.

closed 2024-06-04 by zm711