SpikeInterface/spikeinterface #2957: Importing exporters in windows and mac takes too long

h-mayorquin, 2024-06-01:
Described here:

https://github.com/SpikeInterface/spikeinterface/pull/2955

Checkout the import times (for windows here)
[image]

Maybe someone with mac can figure this one out and hopefully is the same issue that is on windows? 

@zm711 @chrishalcrow

zm711, 2024-06-01:
I can take a look this week! Is there any way to know whether it is report or phy that is the problem? Or maybe the separate common file which needs to make the fixture to start with?

zm711, 2024-06-01:
```python
[In [1]: %timeit import spikeinterface.exporters
The slowest run took 16.75 times longer than the fastest. This could mean that an intermediate result is being cached.
774 ns $ 1.12 us per loop (mean ≥ std. dev. of 7 runs, 1 loop
```

when I ran timeit it this is what it said. I'll try to figure what is going on!

zm711, 2024-06-02:
[image]

So when I run the import script in my terminal though I get reasonable values. (this would mimic the full dependency test for this env).

zm711, 2024-06-03:
By contrast on my Windows system the times are all horrible (running from your PR):

[image]

h-mayorquin, 2024-06-03:
I think we need to run a profile to see where the time is being spent. I can take care of windows.

h-mayorquin, 2024-06-03:
Wait, then let's me just do the profile in the CI then which is the obvious thing to do.

chrishalcrow, 2024-06-04:
> <img alt="Screenshot 2024-06-02 at 7 37 53 PM" width="656" src="https://private-user-images.githubusercontent.com/92116279/335905433-2671c859-30d4-4d43-b8b1-196652dbc782.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0ODY4NDQsIm5iZiI6MTcxNzQ4NjU0NCwicGF0aCI6Ii85MjExNjI3OS8zMzU5MDU0MzMtMjY3MWM4NTktMzBkNC00ZDQzLWI4YjEtMTk2NjUyZGJjNzgyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDA3MzU0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFhZTI1MjA4ZDViOTQ0ZmMxNmMxMDczMzFjOGM3ZDE1MDhkNWExZTM4ODE1OWI4NWJmNTA4MWRjYzZhNTZmNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yPotMVivlJ5edOJ4DerWROc68Id22ENrO4IzHrKnWSk">
> So when I run the import script in my terminal though I get reasonable values. (this would mimic the full dependency test for this env).
> 
> And run from your PR:
> 
> <img alt="Screenshot 2024-06-02 at 7 45 29 PM" width="876" src="https://private-user-images.githubusercontent.com/92116279/335905803-e25a65a5-7bfe-40da-9557-ead69fbab3f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0ODY4NDQsIm5iZiI6MTcxNzQ4NjU0NCwicGF0aCI6Ii85MjExNjI3OS8zMzU5MDU4MDMtZTI1YTY1YTUtN2JmZS00MGRhLTk1NTctZWFkNjlmYmFiM2Y0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDA3MzU0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiOGE1Y2EwNTUwNzgwNmY1ZDJhMjFjYjIyODliY2UwZDFmYjhiY2M2ODYyZThjMzYwYWYxNDNhMjJhMGFjZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mZoEyhlnYNK3skl_q0lhFzSN2NqGanu8hvaRMGEm6zw">

My profiling is in line with these

JoeZiminski, 2024-06-04:
I'm also finding slow imports on Windows in general, this may be unrelated to spikeinterface (some benchmarking below). It seems most of the time is taken importing `scipy`, `sklearn`, `matplotlib`, `numba`. Just profiling import of `scipy.stats` alone takes ~1 second on my windows laptop!

I am running in a conda environment. I have seen some posts suggesting Windows Defender can play a role slowing down imports, but temporarily turning of 'Virus and Threat Protection' did not help (though, Defender was still running, it might still be messing around in the background). I also deleted some paths from my system path in case it affected discovery speed, but it didn't help too much either. I guess the CI is running its macOS and Windows tests within containers on Linux machines, so maybe it doesn't have some of the usual 'additions' that comes with Windows installs.

**spikeinterface.exporters**
[image]

**scipy.stats**
[image]

Was run with:
`python -X importtime -c "import spikeinterface.exporters" 2> exporters.log`
`tuna exporters.log`

zm711, 2024-06-04:
I think that is an excellent summary of the issue and that graphic is awesome. Thanks @JoeZiminski!

h-mayorquin, 2024-06-05:
@JoeZiminski  thanks a bunch.

OK, so how to solve the issue should be clear now. We need to delay the of both scipy stats and matplotlib till executation. Right now there must be some try except blocks that are importing those libraries that are only run when the dependency is installed.

h-mayorquin, 2024-06-05:
Yes, it does:

[image]

https://github.com/SpikeInterface/spikeinterface/actions/runs/9388536782?pr=2983

We can improve this further if you want guys. The extractors is still kind of slow and as I mentioned there is still some try-except blocks for torch that are probably messing up import but not tested in the environment because torch is not in ful (same for hdbscan).

Now the test measure that all the imporst take less than 1.5 secons, maybe we can under a second.

My suggestion would be:
1) merge #2983  
2) Re-profile extractors and full
3) Improve and reduce the import time tolerance to 1 second which I think is a good goal.

zm711, 2024-06-06:
I find this profiling super interesting, but one issue is it is very setup specific. If you look at my Windows profile you see that an import of full takes my computer ~5 seconds, so I just want us to keep in my mind that optimizing for a Github runner is a good first step, but I would love for us to think deeper about how to ensure this gets down to all users too!

h-mayorquin, 2024-06-06:
I think it should depend mainly on the libraries that you have. Could you run Joe command above

```
python -X importtime -c "import spikeinterface.exporters" 2> exporters.log
tuna exporters.lo
``` 

when you have a chance?
I guess you will need to install tuna.

zm711, 2024-06-06:
My comparison was directly between a Mac and with Windows with:

```
pip install .[full,widgets]
```

So they all had the same dependencies. But the wheels would be built differently so I'm more worried about the fact that AMD vs Intel vs M-series will all have slightly different builds for numpy, sci-kit.

If we decide to really pursue this I'm fine installing tuna and doing this, just depends on deep we want to go with this.

h-mayorquin, 2024-06-06:
I just want to see which dependency is the one that is taking so long.  I will try this myself on windows later today as well.

zm711, 2024-06-06:
[image]

Seems like sorters and exporters are the big problems.

h-mayorquin, 2024-06-06:
Can you share the bottom part. This is the one that I want to see. Sorry for being annoyed I will run this myself and is a good idea just to check that everything is there as well.

zm711, 2024-06-06:
[image]

zm711, 2024-06-06:
It's a little too big for me to easily fit everything at once. 

Basically the slow parts are sklearn, scipy and matplotlib and numba.

h-mayorquin, 2024-06-06:
Thanks a bunch.  That plot is very useful. It just allowed me to discover why the extractors are slow at import time. Neo is imported in some places and that in turns imports scipy. I just pushed a fix:

https://github.com/SpikeInterface/spikeinterface/compare/main...h-mayorquin:remove_more_imports?expand=1

You are right that the main problem is scipy, numba and sickit. But I think we got rid of most of them. Are you sure you updated to the latest version that I just merged a couple of hours ago?

The widgets utils should not import matplotlib anymore and spike circus 2 should not import hdbscan. I will check this on my side but I looked for some of the imports that you file indicated and I could not find it.

h-mayorquin, 2024-06-06:
At the end, I think we can get rid of scipy and scikit but for numba we will have to discuss with Sam as that needs a wrapper. But I think there is a lot to gain till we get to that point.

zm711, 2024-06-06:
I had not pulled the latest changes, but just tested your PR #2991 and here is what we have. Cut in half for my windows:
[image]

h-mayorquin, 2024-06-06:
OK, @zm711 , thanks a lot. To me, this confirms my assement. The last frontier is numba indeed and it looks **that scipy is imported through numba!**

 I will close this after the latest https://github.com/SpikeInterface/spikeinterface/pull/2991 is merged.

All the rest is small fry and numba.

PR #2991 (2024-06-10): Remove neo top level imports
Thanks to @zm711 whose profile helped in this [discussion](https://github.com/SpikeInterface/spikeinterface/issues/2957) to find out that import neo `import neo` imports scipy under the hood.

This should make the imports of the extractors module faster.

This should close #2957.
closed 2024-06-10 by alejoe91