SpikeInterface/spikeinterface #2960: Artifact removal average

CGonzC, 2024-06-03:
Hello!

I am trying to remove the light-induced artifact in my recordings with the remove_artifacts function. It works perfectly when "cubic" or "linear" are used. When I try to use the "average" method, it apparently does the average computations, but no changes are observed in the recordings.
[image]

What am I missing?

Thank you very much in advance.

Cristian

CGonzC, 2024-06-03:
HI @alejoe91 ,

Thanks for your quick reply!
They are very similar artifacts per channel. They differ among channels only. Regarding the alignment, they are bound to the light TTL, so I expect no changes there.
 What I get is an unchanged recording, the artifacts are identical when superimposed with the original recording. 
Should I change anything else in the function out of the default?

Thanks again!

Cristian

alejoe91, 2024-06-03:
Mmm that is strange. What version are you using?

@yger any idea what could be wrong?

CGonzC, 2024-06-03:
I'm using SpikeInterface version: 0.100.0.dev0

alejoe91, 2024-06-03:
Can you try to plot the content of:

```
rec_cleaned4._kwargs["artifacts"]
```
This are the estimated average artifacts

CGonzC, 2024-06-03:
It worked after updating SI to version 0.101.0. 

@alejoe91  I am so sorry for wasting your time.

Thank you very much!!!
Cristian

alejoe91, 2024-06-03:
No worries at all! Closing then :)

closed 2024-06-03 by alejoe91

CGonzC, 2024-06-03:
Hi @alejoe91 ,

after the update I got a problem in the sorting part. I am running kilosort3 in Docker . But now I got an error:
TypeError: unsupported operand type(s) for /: 'NoneType' and 'str'

[image]

Before the update all worked perfectly.

What could be the problem?

Thank you very much in advance

Cristian