SpikeInterface/spikeinterface #2967: Error in sorting after SI update: TypeError: unsupported operand type(s) for /: 'NoneType' and 'str'

CGonzC, 2024-06-03:
Hi @alejoe91 ,

After updating SI from 100.0 to 101.0 I got a problem in the sorting. I am running kilosort3 in Docker . I got the error:

TypeError: unsupported operand type(s) for /: 'NoneType' and 'str'

[image]

Before the update this worked perfectly.

What could be the problem?

Thank you very much in advance

Cristian

ttngu207, 2024-06-03:
It's likely the same issue as #2965

zm711, 2024-06-03:
@CGonzC, could you try running from my PR and let me know if that works now with the `folder` argument.

closed 2024-06-04 by alejoe91

CGonzC, 2024-06-04:
It works perfectly in Docker!

Thank you very much @zm711 !!!

PR #2968 (2024-06-04): Patch for regularizing folder arguments
Fixes #2967 , Fixes #2965. 

A bug noticed by @ninelK. 

If folder is None we should set folder to the output for docker/singularity.

To answer your question in the commit history Nina,

they can be strings because we are adding here then we switch to path. We could do the same with a Path, but it doesn't really matter. 

Thanks for spotting the bug. related to #2917.--Needs to be tested by someone with Docker.