SpikeInterface/spikeinterface #2972: TypeError: unsupported operand type(s) for /: 'str' and 'str'

AbhiSwamiUConn, 2024-06-04:
I've recently started to use spike Interface, and so far it's been working well.   I am running into an issue with the detect_peaks method on version 100.6.  Am I doing something wrong?  I notice there are similar issues others have faced such as #2967 but those appear to be for the run sorter method.  
```
#Import modules
import spikeinterface.full as si
import probeinterface as pi
import numpy as np
import spikeinterface.extractors as se
import spikeinterface.preprocessing as st
import spikeinterface.sorters as ss
import spikeinterface.exporters as sx
import spikeinterface.postprocessing as sp
import spikeinterface.sortingcomponents.clustering as ssc
import spikeinterface.sortingcomponents.peak_detection as ssd
import spikeinterface.sortingcomponents.peak_localization as ssl
import spikeinterface.sortingcomponents.peak_selection as sss
import spikeinterface.sortingcomponents.matching as ssm
from probeinterface import generate_multi_shank
from probeinterface.plotting import plot_probe
#Read File
plx_file = "file"
output_folder1 = "output_folder"
output_folder2 = "output_folder"
output_folder3 = "output_folder"
output_folder4 = "output_folder"
#stream id 0 is spike channels; 109,989,908 data points, 40000hz sampling rate
#stream id 1 is wideband channels; 109,989,948 data points, 40000hz sampling rate
#stream id 2 is field potential channels; 27,497,477 data points, 10000hz sampling rate
plx_extractor = se.read_plexon(plx_file, stream_id="1")
print(plx_extractor)
print(plx_extractor.channel_ids)
#Attach probe
probe = generate_multi_shank(num_shank=3, num_columns=1, num_contact_per_column=1)
probe.set_shank_ids(np.arange(3))
probe.set_contact_ids(np.arange(3))
probe.set_device_channel_indices(np.arange(3))
pi.write_probeinterface('my_probe_setup.json', probe)
recording_with_probe = plx_extractor.set_probe(probe, group_mode="by_shank", in_place=True)
print(recording_with_probe.get_probe())
plot_probe(recording_with_probe.get_probe(), with_contact_id = True)
#Filter
recording_f = st.bandpass_filter(recording_with_probe, freq_min=300, freq_max=6000)
recording_cmr = st.common_reference(recording_f, "global")
peaks = ssd.detect_peaks(recording_f, method="by_channel", gather_mode="npy", folder=output_folder4, names=["by_channel"])
#Sort
sorting = ss.run_sorter("spykingcircus2", recording_cmr, output_folder= output_folder1, remove_existing_folder= True)
Sanalyzer = si.waveform_extractor.extract_waveforms(recording=recording_cmr, sorting=sorting, folder = output_folder2, overwrite= True)
#Get information
templates_amplitudes = si.get_template_amplitudes(Sanalyzer, "pos", "extremum")
spike_locations = sp.compute_spike_locations(Sanalyzer)
spike_amplitudes = sp.compute_spike_amplitudes(Sanalyzer)
sx.export_to_phy(Sanalyzer, output_folder=output_folder3, remove_if_exists=True)
```

TypeError                                 Traceback (most recent call last)
Cell In[7], line 4
      2 recording_f = st.bandpass_filter(recording_with_probe, freq_min=300, freq_max=6000)
      3 recording_cmr = st.common_reference(recording_f, "global")
----> 4 peaks = ssd.detect_peaks(recording_f, method="by_channel", gather_mode="npy", folder=output_folder4, names=["by_channel"])

File ~\anaconda3\Lib\site-packages\spikeinterface\sortingcomponents\peak_detection.py:121, in detect_peaks(recording, method, pipeline_nodes, gather_mode, folder, names, **kwargs)
    118             node.parents = [node0] + node.parents
    119         nodes.append(node)
--> 121 outs = run_node_pipeline(
    122     recording,
    123     nodes,
    124     job_kwargs,
    125     job_name=job_name,
    126     gather_mode=gather_mode,
    127     squeeze_output=squeeze_output,
    128     folder=folder,
    129     names=names,
    130 )
    131 return outs

File ~\anaconda3\Lib\site-packages\spikeinterface\core\node_pipeline.py:455, in run_node_pipeline(recording, nodes, job_kwargs, job_name, mp_context, gather_mode, gather_kwargs, squeeze_output, folder, names)
    453     gather_func = GatherToMemory()
    454 elif gather_mode == "npy":
--> 455     gather_func = GatherToNpy(folder, names, **gather_kwargs)
    456 else:
    457     raise ValueError(f"wrong gather_mode : {gather_mode}")

File ~\anaconda3\Lib\site-packages\spikeinterface\core\node_pipeline.py:614, in GatherToNpy.__init__(self, folder, names, npy_header_size, exist_ok)
    612 self.final_shapes = []
    613 for name in names:
--> 614     filename = folder / (name + ".npy")
    615     f = open(filename, "wb+")
    616     f.seek(npy_header_size)

TypeError: unsupported operand type(s) for /: 'str' and 'str'

alejoe91, 2024-06-04:
Hi @AbhiSwamiUConn 

This looks like a bug on our side, where a `str` is not properly converted to path.
Can you try to define your output folder as this?
```
from pathilb import Path

output_folder4 = Path("output_folder")
```

AbhiSwamiUConn, 2024-06-04:
That works.  However, I am getting an error further in the code when I run the spykingcircus2 sorter.  My code is executed but it spends a lot of time creating a new probe when one is already given.  
UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn(\"There is no Probe attached to this recording. Creating a dummy one with contact positions\")

alejoe91, 2024-06-04:
That is not an issue ;) the "problem" is that only channel locations are passed to the sorter, not the full Probe object. But that's everything the sorter needs!

AbhiSwamiUConn, 2024-06-04:
Thank you for your help.

AbhiSwamiUConn, 2024-06-04:
Sorry for commenting in this thread again.  Just for clarification, does spike interface have the ability to filter and detect spikes, from a wideband signal before sorting or do I need to provide the detected spikes?

h-mayorquin, 2024-06-05:
Should be fixed by https://github.com/SpikeInterface/spikeinterface/pull/2982

commit 9152803a (2024-06-06): Merge pull request #2982 from h-mayorquin/fix_path_bug

Fix #2972