SpikeInterface/spikeinterface #2974: Issue with reading intan .rhs file

Fusenheizi, 2024-06-05:
I encountered the following error when trying to convert Intan .rhs files to binary files in Kilosort:

```
TypeError: '>=' not supported between instances of 'int' and 'Version'
```

I then ran this code separately to read the Intan .rhs file and encountered the same issue. 

The source code is as follows:
```
from spikeinterface.extractors import read_intan
# NOTE: You will need to select the appropriate data stream. If you run without
#       specifying `stream_id`, you will get an error message explaining what
#       each stream corresponds to.
filepath = r"F:\cl_exp\rat9_10_11_240403_114532\rat9_10_11_240403_114532.rhs"
recording = read_intan(filepath, stream_id='0')
```

Error message:
```
File ~\.conda\envs\kilosort\lib\site-packages\neo\rawio\intanrawio.py:620, in read_rhs(filename, file_format)
    618 if global_info["notch_filter_mode"] == 2 and global_info["major_version"] >= V("3.0"):
    619     global_info["notch_filter"] = "60Hz"
--> 620 elif global_info["notch_filter_mode"] == 1 and global_info["major_version"] >= V("3.0"):
    621     global_info["notch_filter"] = "50Hz"
    622 else:
```
TypeError: '>=' not supported between instances of 'int' and 'Version'

Fusenheizi, 2024-06-05:
Full error message:

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[3], line 6
      2 # NOTE: You will need to select the appropriate data stream. If you run without
      3 #       specifying `stream_id`, you will get an error message explaining what
      4 #       each stream corresponds to.
      5 filepath = r"[F:\cl_exp\rat9_10_11_240403_114532\rat9_10_11_240403_114532.rhs](file:///F:/cl_exp/rat9_10_11_240403_114532/rat9_10_11_240403_114532.rhs)"
----> 6 recording = read_intan(filepath, stream_id='0')

File [~\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\intan.py:34](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/spikeinterface/extractors/neoextractors/intan.py#line=33), in IntanRecordingExtractor.__init__(self, file_path, stream_id, stream_name, all_annotations, use_names_as_ids)
     32 def __init__(self, file_path, stream_id=None, stream_name=None, all_annotations=False, use_names_as_ids=False):
     33     neo_kwargs = self.map_to_neo_kwargs(file_path)
---> 34     NeoBaseRecordingExtractor.__init__(
     35         self,
     36         stream_id=stream_id,
     37         stream_name=stream_name,
     38         all_annotations=all_annotations,
     39         use_names_as_ids=use_names_as_ids,
     40         **neo_kwargs,
     41     )
     43     self._kwargs.update(dict(file_path=str(Path(file_path).absolute())))

File [~\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:187](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py#line=186), in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    158 def __init__(
    159     self,
    160     stream_id: Optional[str] = None,
   (...)
    165     **neo_kwargs: Dict[str, Any],
    166 ) -> None:
    167     """
    168     Initialize a NeoBaseRecordingExtractor instance.
    169 
   (...)
    184 
    185     """
--> 187     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    189     kwargs = dict(all_annotations=all_annotations)
    190     if block_index is not None:

File [~\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:27](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py#line=26), in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     23 def __init__(self, block_index, **neo_kwargs):
     24 
     25     # Avoids double initiation of the neo reader if it was already done in the __init__ of the child class
     26     if not hasattr(self, "neo_reader"):
---> 27         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     29     if self.neo_reader.block_count() > 1 and block_index is None:
     30         raise Exception(
     31             "This dataset is multi-block. Spikeinterface can load one block at a time. "
     32             "Use 'block_index' to select the block to be loaded."
     33         )

File [~\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:66](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py#line=65), in _NeoBaseExtractor.get_neo_io_reader(cls, raw_class, **neo_kwargs)
     64 neoIOclass = getattr(rawio_module, raw_class)
     65 neo_reader = neoIOclass(**neo_kwargs)
---> 66 neo_reader.parse_header()
     68 return neo_reader

File [~\.conda\envs\kilosort\lib\site-packages\neo\rawio\baserawio.py:189](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/neo/rawio/baserawio.py#line=188), in BaseRawIO.parse_header(self)
    176 """
    177 Parses the header of the file(s) to allow for faster computations
    178 for all other functions
    179 
    180 """
    181 # this must create
    182 # self.header['nb_block']
    183 # self.header['nb_segment']
   (...)
    186 # self.header['spike_channels']
    187 # self.header['event_channels']
--> 189 self._parse_header()
    190 self._check_stream_signal_channel_characteristics()
    191 self.is_header_parsed = True

File [~\.conda\envs\kilosort\lib\site-packages\neo\rawio\intanrawio.py:117](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/neo/rawio/intanrawio.py#line=116), in IntanRawIO._parse_header(self)
    107     else:
    108         self.file_format = "header-attached"
    110     (
    111         self._global_info,
    112         self._ordered_channels,
    113         data_dtype,
    114         header_size,
    115         self._block_size,
    116         channel_number_dict,
--> 117     ) = read_rhs(self.filename, self.file_format)
    119 # 3 possibilities for rhd files, one combines the header and the data in the same file with suffix `rhd` while
    120 # the other two separates the data from the header which is always called `info.rhd`
    121 # attached to the actual binary file with data
    122 elif self.filename.endswith(".rhd"):

File [~\.conda\envs\kilosort\lib\site-packages\neo\rawio\intanrawio.py:620](http://localhost:8888/lab/tree/~/.conda/envs/kilosort/lib/site-packages/neo/rawio/intanrawio.py#line=619), in read_rhs(filename, file_format)
    618 if global_info["notch_filter_mode"] == 2 and global_info["major_version"] >= V("3.0"):
    619     global_info["notch_filter"] = "60Hz"
--> 620 elif global_info["notch_filter_mode"] == 1 and global_info["major_version"] >= V("3.0"):
    621     global_info["notch_filter"] = "50Hz"
    622 else:

TypeError: '>=' not supported between instances of 'int' and 'Version'

zm711, 2024-06-05:
I just patched this on the Neo side. Thanks for finding this!

zm711, 2024-06-05:
Patch is now merged. Install from there and that should fix it!

zm711, 2024-06-05:
I'll close this now, but feel free to open fresh issues if they come up!

closed 2024-06-05 by zm711