SpikeInterface/spikeinterface #2985: feature request: bad channel exclusion from common_reference function

WeissShahaf, 2024-06-06:
Dear devs,
I appreciate your hard work. spikeinterface is becoming a really great system for nerual data processing. 

Please consider implementing a bad channel exclusion option for common_reference function.
similiar to catGT's -chnexcl={prb;chans}     ;this probe, exclude listed chans from ap loccar, gblcar, gfix

maybe something like this:
si.common_reference(rec, operator="median",chnexcl=bad_channel_ids)

Thank you,
Shahaf

alejoe91, 2024-06-06:
Hi @WeissShahaf 

Thanks for the support :)

You can already do that! If you have a list of channels to exclude, you can use the `remove_channels()` function:
```
bad_channel_ids = [...+

recording_clean = recording.remove_channels(bad_channel_ids)
recording_clean_cmr = spre.common_reference(recording_clean)
```
Is this what you had in mind?

WeissShahaf, 2024-06-06:
@alejoe91 
no, the functionality i'm describing is to exclude channels from the CMR function, but keep them in the recording.

h-mayorquin, 2024-06-06:
Does this accomplish what you want: @WeissShahaf 

```python
from spikeinterface.core import generate_recording, aggregate_channels
from spikeinterface.preprocessing import common_reference

recording = generate_recording(num_channels=4, durations=[10], seed=0)
recording = recording.rename_channels(new_channel_ids=["a", "b", "c", "d"])
bad_channels = ["b"]
recording_for_crm = recording.remove_channels(remove_channel_ids=bad_channels)
recordin_bad_chanenls = recording.select_channels(channel_ids=bad_channels)

common_reference = common_reference(recording_for_crm, reference='global')

recording_where_some_channels_are_excluded_from_crm = aggregate_channels([common_reference, recordin_bad_chanenls])
recording_where_some_channels_are_excluded_from_crm.get_traces()

```

alejoe91, 2024-07-04:
@WeissShahaf fixed here #3139, but with a slight different design:

You can optionally pass a list of channels to compute the median on with the `ref_channel_ids` existing argument

h-mayorquin, 2024-07-04:
To be more precise you would do

```python
from spikeinterface.core import generate_recording
from spikeinterface.preprocessing import common_reference

recording = generate_recording(num_channels=4, durations=[10], seed=0)
recording = recording.rename_channels(new_channel_ids=["a", "b", "c", "d"])
good_channels = ["a", "b", "c"]

common_reference = common_reference(recording_for_crm, reference='global',  ref_channel_ids=good_channels)
```

I am closing this as soon as  merge https://github.com/SpikeInterface/spikeinterface/pull/3139 if no one disagrees.

closed 2024-07-08 by alejoe91