SpikeInterface/spikeinterface #2986: Is there a widget for plotting the 'peak localisation'?

JoeZiminski, 2024-06-06:
In the handle motion / drift tutorial, [there is a nice peak localisation plot](https://spikeinterface.readthedocs.io/en/stable/how_to/handle_drift.html#plot-peak-localization) that is coded up in the tutorial. I think I also saw this at the workshop. Is this implemented in a widget? If not and people agree it's a good addition, happy to open a PR!

zm711, 2024-06-06:
This might be it:

https://github.com/SpikeInterface/spikeinterface/blob/a8ed7c8172e11dfd25d716408bc0cd48049dac44/src/spikeinterface/widgets/motion.py#L10

alejoe91, 2024-06-06:
I think we should split that `plot_motion` widgets. I find it very convenient to plot the rastermaps alone and tweak plotting params

JoeZiminski, 2024-06-06:
Thanks both! I think that `plot_motion` plots this:
[image]

As far as I know it cannot plot this?

[image]

That makes sense to split in some way, maybe it could take an argument to only plot one of the panels if preferred?

samuelgarcia, 2024-06-06:
I will refator plot_motion along side the Motion obect refactoring.

JoeZiminski, 2024-06-06:
@samuelgarcia is it correct that there is currently no widget for the peak localisation over the probe plots? If so would it make sense for me to add one?

closed 2024-06-21 by alejoe91

JoeZiminski, 2024-06-21:
The tests for this widget will be resurrected after the big `widgets` refactor, can be found [here](https://github.com/SpikeInterface/spikeinterface/pull/3022/commits/7ab068b36bcc55d1efd6966051f54b152c4321e2)

PR #3022 (2024-06-21): Add peaks_on_probe widget.
This PR adds a 'Peaks on Probe' widget that plots the peak locations overlaid on a plot of the probe (I am not a huge fan of the name myself, feel free to suggest others!). This PR closes #2986.

[image]

This plot has come up a few times in [documentation](https://www.google.com/search?q=handle+motion%2Fdrift+spikeinterface&oq=handle+motion%2Fdrift+spikeinterface&gs_lcrp=EgZjaHJvbWUyBggAEEUYOTIGCAEQRRg8MgYIAhBFGDwyBggDEEUYPNIBCDg3NTBqMGo3qAIAsAIA&sourceid=chrome&ie=UTF-8) and [tutorials](https://spikeinterface.github.io/blog/) so I think could be a handy plotting fu [... cut]