SpikeInterface/spikeinterface #2994: Why does the spike amplitude shrink to single digits after preprocessing?

BEGINRX, 2024-06-07:
`recording_cmr = si.common_reference(raw_rec, reference='global', operator='median', dtype='float32')
recording_f = si.bandpass_filter(recording_cmr, freq_min=filting['freq_min'], freq_max=6000, dtype='float32')
rec_w = si.whiten(recording=recording_f, apply_mean=True)
rec_corrected = si.correct_motion(recording=rec_w, preset='kilosort_like', detect_kwargs=detection)`

Is this preprocessing process correct? Why is the amplitude in single digits after whitening?

JoeZiminski, 2024-06-07:
Hi @BEGINRX, I believe this is the expected behaviour of whitening. Whitening performs (I would say quite an agressive) operation on the data to remove correlations between channels. You can see the effect on [slide 24 of these slides](https://docs.google.com/presentation/d/1qS-Ua1qbegiHr-_wCDjGk1dW6ydw09YYdpP0FJpilcQ/edit#slide=id.g1e96a3a1718_0_16). In this case the 'correlation' it removes is largely the APs, which end up loading strongly only onto a subset of their original channels.

BTW It is reccomended to to apply whitening after motion correction, the order I would suggest is:

1) bandpass filter,
2) CMR
3) correct motion
4) whiten

BEGINRX, 2024-06-07:
Thank you for your reply. But I still have some problems. 
This is the raw recording after "get_traces". The amplitude range is between 1000 to -1000. 
[image]
But after whiten as your advice, the amplitude range is between 15 to -15.
[image]
The code is 
`recording_cmr = si.common_reference(raw_rec, reference='global', operator='median', dtype='float32')`
`recording_f = si.bandpass_filter(recording_cmr, freq_min=filting['freq_min'], freq_max=6000, dtype='float32')`
`rec_corrected = si.correct_motion(recording=recording_f, preset='kilosort_like', detect_kwargs=detection)`
`rec_w = si.whiten(recording=recording_f, mode='local')`
`trace = rec_w.get_traces()`
And the amplitude influences the waveforms. So how can I do?

alejoe91, 2024-06-07:
@BEGINRX That's what whitening does. If you want "real" waveforms, compute them prior to whitening :)

BEGINRX, 2024-06-07:
OK，I see. So if I want to extract waveforms from recordings using function "extract_waveforms", do I need to input the raw recording?

JoeZiminski, 2024-06-07:
I would suggest inputting the motion-corrected recording, so the data will have under gone all preprocessing steps up to but not including whitening. If you put the raw recording, it may be noising / subject to motion artefacts etc. Let us know if it still does not look right!

zm711, 2024-06-19:
I'll close this one, but if more questions come up feel free to open new issues!

closed 2024-06-19 by zm711